riscv/riscv-isa-manual #1702: When the PM feature is enabled, can PMLEN be set to 7 in sv48 mode?

chara811, 2024-10-30:
Regarding sv48 mode, there is no clear restriction on setting PMLEN to 7 in the PM specification

gfavor, 2024-10-30:
Yes.  The PMM fields are WARL and nothing in the spec restricts which PMLEN values can be used with which translation modes.

chara811, 2024-11-04:
Does PM function support when *atp=sv39?

chara811, 2024-11-04:
The PM spec Section 3.5 describes configurations such as "PMLEN=7 and hgatp.MODE=sv57x4,PMLEN=16 and hgatp.MODE=sv57x4,"PMLEN=16 and hgatp.MODE=sv48x4." However, it does not list whether sv39x4 can be configured with PMLEN=16 or 7, and sv48x4 can be configured with PMLEN=7.Are these configurations that are not listed supported?

gfavor, 2024-11-04:
That spec text does not limit support to the listed configs.  That text is specifying when one would need to execute an HFENCE.GVMA when changing henvcfg.PMM (and implicitly is specifying when an HFENCE.GVMA is not necessary).