riscv/riscv-isa-manual #1704: When stap =bare, can we configure vsatp and hgatp !=bare for two-stage address translation ?

chara811, 2024-11-01:


gfavor, 2024-11-01:
satp controls S/HS-mode translation.  vsatp and hgatp control two-stage transation.  So they are quite independent of each other.  Plus, in any case, all three can be configured independently of each other (the architecture expresses no constraints on this).

rock-ifly, 2024-11-01:
> satp controls S/HS-mode translation. vsatp and hgatp control two-stage transation. So they are quite independent of each other. Plus, in any case, all three can be configured independently of each other (the architecture expresses no constraints on this).

Does this mean that, when HostOS doesn't use the MMU(satp.mode=Bare),  the hypervisor and guestOS can still work correctly if vsatp.mode != Bare and hgatp.mode != Bare ?

aswaterman, 2024-11-01:
Yes.

closed 2024-11-01 by aswaterman