riscv/riscv-isa-manual #1730: In the description of the Smcntrpmf extension in privilege 1.13, it states that exceptional instructions are not considered to retire, but in the privilege description of the minstret register, only ECALL and EBREAK do not increment it.

Mei-x-l, 2024-11-18:
[image]
[image]
When implementing the Smcntrpmf extension, is it correct to understand that all synchronous exceptions in the inhibited mode are not considered to retire, while if the Smcntrpmf is not implemented, only ECALL and EBREAK do not increment the minstret register? Is this understanding correct?

gfavor, 2024-11-18:
No.  In general, if an instruction (including ECALL and EBREAK) is faulted such that it is architecturally considered to have not fully completed execution architecturally (aka retired), then it must not be counted by minstret.  Conversely, if an instruction - including *RET - successfully completes architecturally (and is hence considered to have retired) then it must be counted by minstret.

This is all consistent with the various spec excerpts in the first post.

noionion, 2024-11-19:
Why is it necessary to specifically mention `ecall`/`ebreak` in the privileged specification? When not considering control flow changes caused by the exception service routine, such as `epc` and whether instructions are re-executed, `ecall`/`ebreak` should behave similarly to other instructions that may trigger synchronous exceptions. Can I assume that, in the absence of the Smcntrpmf extension, instructions that cause other exceptions can still be counted by `minstret` before entering the exception service routine? This also seems like a valid implementation according to the spec.

In the example provided in the Smcntrpmf extension, it is stated:
> Only once the load is re-executed and retires will it increment instret.

A re-executed instruction can be treated as a new instruction that executes without exception. However, instructions like ecall/ebreak typically are not re-executed (due to the update to epc), so there is no need for special mention in the privileged specification (as they essentially exhibit the same behavior).

gfavor, 2024-11-19:
ECALL/EBREAK are highlighted by the spec since in one sense they successfully execute insofar as causing a trap to a handler at a higher privilege level.  But they set epc to the address of the instruction - like normal exceptions on an instruction - and not to the address of the next instruction.  And they cause synchronous exceptions and hence are considered to not retire.

In short, ECALL/EBREAK cause synchronous exceptions.  And like other synchronous exceptions, epc is set to the address of the instruction and the instruction does not retire.  But ECALL/EBREAK are highlighted since they do successfully do what they are specifically defined to do (cause an exception), and hence someone might mistakenly consider them to retire (even though this behavior would be inconsistent with all other cases of an instruction causing a synchronous exception).

Mei-x-l, 2024-11-19:
Thank you for your answer. I think I should have understood it.

closed 2024-11-19 by Mei-x-l

pdonahue-ventana, 2024-11-19:
I complained 5+ years ago that "retire" isn't really defined: https://lists.riscv.org/g/tech-debug-archive/topic/32209057#msg800

> Instruction retirement is a key concept in the trace architecture so it mentions it 80 times but the term is still not defined anywhere (description of minstret behavior on two instructions notwithstanding).

Although the ecall/ebreak description was helpful, I still don't think that "retire" is defined.  I guess that we have all figured it out, though.