riscv/riscv-isa-manual #1735: Can it support misaligned memory access when PBMT=NC?

Maxpicca-Li, 2024-11-22:
The Svpbmt mentions misaligned memory access when PBMT=IO.
> Implementations may override additional PMAs not explicitly listed in Table 27. For example, to be consistent with the characteristics of a typical I/O region, a misaligned memory access to a page with PBMT=IO might raise an exception, even if the underlying region were main memory and the same access would have succeeded for PBMT=PMA.

But when PBMT = NC, can it support misaligned memory access with PMA=main_mem or PMA=IO?

Maxpicca-Li, 2024-11-22:
> PBMT= NC or IO is only guaranteed to override the attributes specified in table 27. The following sentence then says "Implementations may override additional PMAs not explicitly listed in Table 27". Note "may".
> 
> In the case of PBMT=NC on top of PMA=main memory, one typically would not override the misaligned behavior for main memory since "NC" is intended to represent noncacheable main memory, and hence misaligned accesses would be supported. In the case of PBMT=NC on top of PMA=IO, and similar to the example in the spec, one typically would override the misaligned behavior to be consistent with the characteristics of main memory.

Thank you for your reply. I still have doubts: It is acceptable to report an exception(PBMT=IO) where there is no exception(PMA=main_mem), but if there is no exception(PBMT=NC) in the place where the exception should be reported(PMA=IO), will this affect the correctness?

gfavor, 2024-11-22:
I wouldn't view it that way.  In each case the PBMT memory type is overriding the memory type implied by the PMAs and is generally intended to be the effective memory type.  So then, from the perspective of the effective memory type, one can then talk about whether misaligned accesses should cause an exception or not (i.e. yes for PBMT=IO and no for PBMT=NC).

Maxpicca-Li, 2024-11-25:
I got. Thank you again!

closed 2024-11-25 by Maxpicca-Li