riscv/riscv-isa-manual #1741: Does riscv allow vector load/store access the same device address multiple times?

AlexGJL, 2024-11-25:
Does riscv allow vector load/store access the same device address multiple times like arm? it seems not easy to deal exception for hardware if not allowed.

gfavor, 2024-11-25:
See the "Precise vector traps" sub-section in the "Exception Handling" of the RVV spec - where it talks about what is and isn't allowed with respect to non-idempotent regions.

aswaterman, 2024-11-26:
Note also that RVV is a bit different than SVE in this regard: RISC-V has the `vstart` CSR, so can restart an instruction partway through at the position of the exception, whereas SVE does not have an equivalent feature and so must restart at the beginning.

AlexGJL, 2024-12-18:
ok, thank you

closed 2025-01-03 by AlexGJL