riscv/riscv-isa-manual #1748: The effect of FENCE on A/D bit update.

zhuotianshu, 2024-12-02:
In Section10.3.1, it says:

> The ordering on loads and stores provided by FENCE instructions and the acquire/release bits on atomic instructions also orders the PTE updates associated with those loads and stores as observed by remote harts.

But it also says:

> The PTE updates due to memory accesses ordered-after a FENCE are not themselves ordered by the FENCE.

Are they conflict with each other?

aswaterman, 2024-12-03:
I think the first sentence is talking about the case where the load/store is before the FENCE, and the second sentence is talking about the case where the load/store is after the FENCE, so there is no contradiction.  (Does that sound right to you, @gfavor? I'm having to page this information back in, so to speak...)

zhuotianshu, 2024-12-03:
Thank you very much, I'll close this issue.

closed 2024-12-03 by zhuotianshu