riscv/riscv-isa-manual #1751: About local order of A bit update.

zhuotianshu, 2024-12-03:
In section 10.3.1:

> Updates of the A bit may be performed as a result of speculation, even if the associated memory access ultimately is not performed architecturally. However, updates to the D bit, resulting from an explicit store, must be exact (i.e., non-speculative), and observed in program order by the local hart.

It specifies that D bit must be oberved in program order, which means that all subsequent loads by the local HART must see this update.

My question is, should the A bit update also obey this program order? In other words, must all subsequent loads by the local HART observe the A bit update caused by a page table walk corresponding to a previous load/store?

gfavor, 2024-12-03:
How do speculative A bit updates (which are implicit memory accesses) represent a violation of the first rule about loads and stores (which are explicit memory accesses), especially since rule 1 is only about D bit updates?  Or are you just observing that A bit updates aren't subject to a similar rule as rule 1 for D bit updates?  (Which doesn't mean that rule 1 for D bit updates can be violated.)

As far as rule 2 for D bit updates, it must not be violated (otherwise it wouldn't actually be a rule or requirement).

zhuotianshu, 2024-12-03:
> Or are you just observing that A bit updates aren't subject to a similar rule as rule 1 for D bit updates? (Which doesn't mean that rule 1 for D bit updates can be violated.)

Yes, that's what I meant. My expression was incorrect.

gfavor, 2024-12-03:
Since A bit updates are speculative, they are unordered wrt other surrounding loads/stores.

zhuotianshu, 2024-12-03:
OK, thank you very much!

closed 2024-12-03 by zhuotianshu