riscv/riscv-isa-manual #1752: About the ordering rule between A/D bit update and subsequent memory access.

zhuotianshu, 2024-12-03:
In section 10.3.1:

> The PTE update must appear in the global memory order before the memory access that caused the PTE update and before any subsequent explicit memory access to that virtual page by the local hart.

As I understand it, this rule is to order the A/D bit update and all subsequent memory access that "would cause such an A/D bit update". But this rule would also prohibit a load to perform when the corresponding PTE already has A bit set, but has D bit unset, and there is a D bit update to be done. Can I just let this load to perform speculatively, not waiting until the D bit being set?

gfavor, 2024-12-03:
"No" since, as the latest spec notes, "D bit updates are still required to be exact and performed in program order, as the D bit affects the functional correctness of page eviction."  But that applies the apparent ordering of these memory accesses.  Just as is generally true about loads and stores, they can speculatively exceute out-of-order wrt each other - as long as they APPEAR to comply with all the architectural ordering rules.

So one could potentially speculatively execute the load before the D bit update, but then one has to consider all the possible weird and/or non-obvious ways that this order could possibly be observed (and make sure that there is no such possible way to observe the two accesses having been physically performed in a different order than what is architecturally required).

So either do the D bit update before performing the load or be extremely thorough and hope you don't miss a case whether the reversed order can become architecturally visible.  (I believe most implementations of hardware D bit updates will do the former.  And keep in mind that these should be rare and hence not a performance issue.)

zhuotianshu, 2024-12-03:
Okay, I understand now. Thank you very much!