riscv/riscv-isa-manual #1753: How to understand the "convert an invalid address into some other invalid address" in 3.1.14 MEPC register description

rock-ifly, 2024-12-03:
How to understand the "convert from an invlid address into some other invalid address"? why the converted address is also called invalid?

Does this relate to address size problem? that's to say for a PAW(PA width)=VAW(VA width) =48 hart, if pc=0xFFFF_ABCD_0000_0000，exception raise， then mepc should save a 48-bit PC( 0xABCD_0000_0000)， not the 0xFFFF_ABCD_0000_0000?

> mepc is a WARL register that must be able to hold all valid virtual addresses. It need not be capable of
holding all possible invalid addresses. Prior to writing mepc , implementations may convert an invalid
address into some other invalid address that mepc is capable of holding.

gfavor, 2024-12-03:
You should (must) not convert an invalid address into a valid address.  But you can convert an invalid address tha tis not representable in mepc to another invalid address that IS representable in mepc.  This allows one to not have to represent all possible 64-bit address values if the largest supported valid address is less than 64 bits.

aswaterman, 2024-12-04:
Or maybe the max valid address size plus one extra bit, so that it's straightforward to synthesize a different bad address.  But yes, you have the right idea.

aswaterman, 2025-12-09:
Closed as answered.