riscv/riscv-isa-manual #1765: What va range should sfence.vma rs1!=0 invalid from TLB?

GuoShibo-cn, 2024-12-11:
sfence.vma instruction with rs1!=0,needs to guarantee reads and writes made to leaf page table entries corresponding to the specific virtual address in rs1 been synchronized.  Is it also necessary to make sure that all virtual address memory mapping within the same page (superpage) also been  synchronized?  Given the fact that page size cached in TLB could be shrank for multiple reasons, i. e. , pmp entry size or hypervisor g stage page size.

aswaterman, 2024-12-11:
Superpage entries _are_ leaf entries.  ("Leaf" doesn't mean 4 KiB; it means whatever the terminal node in the tree is.)  So, yes, if the VA corresponds to a superpage, then all translations associated with that superpage are synchronized by that SFENCE.VMA instruction.

You're of course right that there's some extra complexity to handle this situation with respect to translations that are splintered due to G-stage translation, PMP, or PMA, but that's the requirement.

GuoShibo-cn, 2024-12-12:
Ok,thanks for your answer! it's clear to me.

closed 2024-12-12 by GuoShibo-cn