riscv/riscv-isa-manual #1774: Clarification about access to physical address which is higher than the maximal supported physical address

leBsky, 2024-12-14:
Assuming we have a RISC-V CPU whose maximal supported physical address is 0xff_ffff_ffff (i.e. the maximal supported physical address is **40** bit)
Then what is the behavibor when the following condition occurs, the Spec seems not give a clear definition.
1. M mode or MMU disabled, jump instruction  or load/store instruction which accesses to an address higher than the maximal supported physical address, for example, jump to physical address 0xffff_ffff_fffff. Should it raise an exception? If yes, which exception?
2. S/U mode and MMU enabled, the pte[53:10] PPN filed is 0xfff_ffff_ffff. That is,  page table walk finds a pte which has ppn is higher than maximal physical address 28 bits(40-12). Should it raise an exception? If yes, which exception?

pdonahue-ventana, 2024-12-14:
It sounds like your implementation has a PMA region with no read/write/execute access (previously called a vacant PMA region) starting at address 0x10000000000.  Accessing that region will cause a PMA violation which will manifest as an access-fault exception (or possibly a bus-error interrupt) as described in the PMA section of the spec.

kersten1, 2025-01-09:
@leBsky did that answer your question? If so, can you please close issue. Else maybe add more detail?

closed 2025-01-10 by leBsky