riscv/riscv-isa-manual #1775: Does a software-injected interrupt get affected by the xideleg register

tom-guo-sudo, 2024-12-15:
1.The Privileged Architecture said “SEIP is writable in mip, and may be written by M-mode software to indicate to S-mode that an external interrupt is pending”, is that interrupt could be controlled by mideleg.SEIP? 
2. software can also inject s-level external interrupt by setting mvien=1, mvip[9]=1 and mideleg[9]=0,  so the virtual interrupt will trap to M mode first?

gfavor, 2024-12-16:
Mideleg doesn't control interrupts themselves.  It only controls the delegation of them.  Control of SEIP itself via mip, sip, and an external interrupt controller are all described in various parts of the Priv spec.

The beginning of section 5.3 of the AIA spec I believe answers the second question.

tom-guo-sudo, 2024-12-20:
From section 5.3 of the AIA spec -- "Interrupt filtering permits a supervisor-level interrupt (SEI or SSI) or local or custom interrupt to trap to M-mode and then be selectively delegated by software to supervisor level, even while the corresponding bit in mideleg remains zero."

For SEI interrupt , mideleg[SEI]=0, sie[SEI]=1. M-level Assert SET to s-level . Is the following procedure correct?
1. Current privilege level is M, set mvien[SEI]=1, mvip[SEI]=1. So sip[SEI]=1, sie[SEI]=1
2. The SEI interrupt is s-level interrupt, so interrupt can taken when privilege level return to S mode
3. The interrupt SEI will trap to M mode because mideleg[SEI]=0
4. Interrupt handler can delegate the interrupt to s level

I have another question:
I want to know why interrupt can delegate to s-level when mideleg[SEI]=0.
Does the interrupt context start processing in M-mode and then continue processing in S-mode?

gfavor, 2024-12-20:
Yes, I believe the procedure is correct.  And as the spec in section 5.3 describes, interrupt handling starts in M-mode and then the M-mode handler chooses to pass down (aka delegate) the interrupt to the S-mode interrupt handler - unbeknownst to the latter handler which just understands that it received a supervisor interrupt.

allenjbaum, 2024-12-20:
I think the distinction here is that the Mmode handler is delegating (in
SW), whereas mideleg makes the HW do it.
You can do this even in a system that doesn't have AIA.

tom-guo-sudo, 2024-12-27:
I got it, thanks for your replay

closed 2024-12-27 by tom-guo-sudo