riscv/riscv-isa-manual #1776: Smstateen bit encoding not aligned with ratified specifications.

avpatel, 2024-12-16:
The bit encoding in "4.2. State Enable 0 Registers" does not align with:

1. The ratified Smstateen spec define BIT[63] for next level xstateen CSRs and BIT[62] for xenvcfg CSRs (Refer, 
2. The ratified Ssaia and Smaia defines BIT[60] for indirect CSRs, BIT[59] for AIA, and BIT[58] for IMSIC

Please see attached PDFs which are original ratified Smstateen and AIA specifications.
[riscv-interrupts-1.0.pdf](https://github.com/user-attachments/files/18152675/riscv-interrupts-1.0.pdf)
[Smstateen.pdf](https://github.com/user-attachments/files/18152683/Smstateen.pdf)

ved-rivos, 2024-12-16:
I see this:
[image]

avpatel, 2024-12-16:
I saw this in Nov 28 PDF (https://github.com/riscv/riscv-isa-manual/releases/tag/riscv-isa-release-f3f8a93-2024-11-28)

Looks like this is fixed very recently.