riscv/riscv-isa-manual #1779: Dose any feature for trigger is optional?

AlexGJL, 2024-12-18:
Dose hardware can choose which feature to be implemented for trigger?
 e.g. Can hardware just implement load/store address trigger,  not implement data trigger?
        Can hardware not implement tinfo, mcontext, scontext. etc.

pdonahue-ventana, 2024-12-18:
This is a better question for https://github.com/riscv/riscv-debug-spec/ but I can answer it here.

For WARL fields, the normal WARL rules apply.  It's legal (and common) to implement a subset of mcontrol6 functionality.

The spec is pretty clear about what CSRs are mandatory vs. optional.  For instance:
- mcontext: "This register must be implemented if hcontext is implemented, and is optional otherwise."
- scontext: "This optional register is..."
- tinfo: "This register is optional if... [condition]" and it's mandatory otherwise.

AlexGJL, 2024-12-20:
I get it, thank you

closed 2024-12-20 by AlexGJL