riscv/riscv-isa-manual #1785: Conflict Between shlcofideleg Extension and AIA Specification Regarding Interrupt 13 (locfi) Handling

noionion, 2025-01-06:
It is noted that the shlcofideleg extension places locfi under the VS state, which seems to contradict the AIA specification. The AIA document mentions:

> Table 5.4: The effects of mideleg and mvien on sip and sie (except for the hypervisor extension’s VS-level interrupts, which appear in hip and hie instead of sip and sie). A bit in mvien can be set to 1 only for major interrupts 1, 9, and 13–63. For interrupts 0–12, aliases in sip may be read-only, as specified by the RISC-V Privileged Architecture.

Locfi (interrupt 13) should therefore be classified under sip, but this extension implies that it is possible to set hideleg[13] and reflect it in hip. If this extension is supported, it means that locfi is included within the hip range, and mideleg[13] must also be read-only 1, which could potentially lead to numerous compatibility issues between software and hardware...

gfavor, 2025-01-08:
The shlcofideleg extension depends on or layers on top of the AIA extensions, overriding what is said in AIA.  And so the shlcofideleg extension means what it literally says (as override to what AIA says that may seem to be conflicting when only AIA is implemented).

noionion, 2025-01-08:
According to the description in _riscv-privileged-priv_1_13_:

> The Shlcofideleg extension supports delegating LCOFI interrupts to VS-mode. If the Shlcofideleg extension is implemented, hideleg bit 13 is writable; otherwise, it is read-only zero. When bit 13 of hideleg is zero, vsip.LCOFIP and vsie.LCOFIE are read-only zeros. Otherwise, vsip.LCOFIP and vsie.LCOFIE are aliases of sip.LCOFIP and sie.LCOFIE.

This means that vsip[13] and vsie[13] will be aliases of sip[13] and sie[13], while hie[13] and hip[13] will be read-only zeros. Meanwhile, both mideleg[13] and hideleg[13] are writable.

Considering the possibility of future higher-bit interrupt extensions (perhaps with interrupts exclusive to vs-mode), the hardware design will need to individually alias each bit from 13 to 63 (aliasing to sip or hip)...

It may be necessary to update the AIA documentation to avoid errors or confusing interpretations.

gfavor, 2025-01-08:
The behavior for bits 14 to 63 remain as specified by AIA until an arch extension comes along that overrides the default AIA behavior.  Since any such arch extension lies in the future, this is not something that can be anticipated and spec'ed by AIA.

Also note that support for AIA does not mean that any, some, or all future extensions that introduce new functionality for bits 13 to 63 are also supported.  And AIA, as with all ratified specs, cannot be semantically changed.  It is up to new ratified extensions to modify previously ratified semantic behaviors.

From an interpretation perspective, an implementer takes the AIA spec and then "layers" on top whatever other AIA-related specs that the implementer chooses to also support.

noionion, 2025-01-08:
Thank you for your reply. This issue will be close.

closed 2025-01-08 by noionion

kdockser, 2025-02-22:
From: https://lists.riscv.org/g/tech-privileged/message/2278
### ARC minutes - 2025-02-18

The ARC reviewed a github issue (https://github.com/riscv/riscv-isa-manual/issues/1785) concerning a conflict between the Shlcofideleg and AIA extensions and LCOFI. After discussion, the ARC concluded that Shlcofideleg is an update to the base privileged specification and AIA is an extension from the base, **so that when AIA is implemented, it overrides the base.** The ARC suggests that the AIA specification be updated to be clearer about it superseding the base privileged specification in this area.