riscv/riscv-isa-manual #1795: Clarification Needed: Concerns Regarding PMP Changes Without HFENCE.VVMA Usage

Wangrodman, 2025-01-10:
[image]

HFENCE.VVMA targets VS-stage translation, while HFENCE.GVMA targets G-stage translation.

I have concerns about the above description regarding PMP changes. PTE caching is a common practice in MMUs, meaning that PTEs accessed implicitly during VS-stage translation can be cached. If the PMP settings for the address of a cached PTE are changed, the new settings will not be visible to the cached entry.

As a result, the expected trap will not occur because the cached PTE is used without checking the new settings.
In this case, whether a fault should occur is undefined, correct? If the software requires consistent and predictable protection, HFENCE.VVMA should be executed. This responsibility should be documented in the software guidelines by whoever provides a product that includes the PTE cache.

aswaterman, 2025-01-10:
The spec clearly states that HFENCE.GVMA suffices and HFENCE.VVMA is not required.  There is no problem here.

> HFENCE.VVMA targets VS-stage translation, while HFENCE.GVMA targets G-stage translation.

This is not entirely correct.  As noted above, HFENCE.GVMA suffices in this case, meaning that in certain cases it is more powerful than you presume.

Wangrodman, 2025-01-10:
> > HFENCE.VVMA targets VS-stage translation, while HFENCE.GVMA targets G-stage translation.
> 
> This is not entirely correct. As noted above, HFENCE.GVMA suffices in this case, meaning that in certain cases it is more powerful than you presume.

Do you mean that implicit G-stage translation is also targeted by HFENCE.GVMA?
If so, it makes sense that GVMA is sufficient.

aswaterman, 2025-01-10:
> Do you mean that implicit G-stage translation is also targeted by HFENCE.GVMA?

No.  I only mean to say that the spec clearly states that this particular case is covered by HFENCE.GVMA.  I am not making a broader statement.

aswaterman, 2025-01-10:
Closing this issue, since the spec is crystal clear.

terranfund, 2025-01-13:
Hi @aswaterman 
[image]
Since HFENCE.GVMA won't flush the VS-stage address translation, and the data collected along with it, including PMP-check encountered wrt implicit access in support for VS-stage address translation and the G-stage address translation that accesses are subject to.
For explicit access, RISC-V SPEC is complete.
However, for such implicit access, HFENCE.VVHA should be involved to synchronize though.

gfavor, 2025-01-13:
As Andrew stated earlier , "_The spec clearly states that HFENCE.GVMA suffices and HFENCE.VVMA is not required._"  The spec excerpt in the preceding post is consistent with this.

IF an implementation caches PMP information corresponding to the final SPA as part of caching VS-stage translations, then that cache is subject to HFENCE.GVMA's (as the spec says).  If an implementation only caches the GPA result of a VS-stage translation (and associated VS-stage permissions), then that cache is not subject to HFENCE.GVMA's (even though the implicit PTE accesses to do the VS-stage tablewalk had to each be translated/etc. by implicit G-stage translations and PMPs/PMAs).

terranfund, 2025-01-13:
> If an implementation only caches the GPA result of a VS-stage translation (and associated VS-stage permissions), then that cache is not subject to HFENCE.GVMA's (even though the implicit PTE [...]
I agreed. Since HFENCE.GVMA cannot handle this part, can HFENCE.VVMA handle this or say, by SPEC, VVMA naturally can handle this.

terranfund, 2025-01-13:
> If an implementation only caches the GPA result of a VS-stage translation (and associated VS-stage permissions), then that cache is not subject to HFENCE.GVMA's (even though the implicit PTE [...]
I agreed. Since HFENCE.GVMA cannot handle this part, can HFENCE.VVMA handle this or say, by SPEC, VVMA natually can handle this.