riscv/riscv-isa-manual #1800: the mode to take LCOFI once it's delegated

Steven-Li-Xiaogang, 2025-01-13:
Once we support Sscofpmf extension, the local counter overflow interrupt should be supported, also it can be delegated to S-mode and VS-mode. For hardware implementation, the S-mode and VS-mode accessiable CSRs hpmcounter and hpmevent are shadow of mhpmcounter and mhpmevent, then it's not clear describe how should the interrupt be raised/taken.

1. The LCOIP bit is not defined as Timer/Software/External interrupt, which have separate pending bits and enable bits for different mode, so should there only one pending register for LCOFIP in hole system, all privileges access LCOFIP is targeted to the same instance?
2. If there is only one instance, which mode should the LCOFIP be taken into
   1 ) M-mode only  or
   2) the mode and only the mode according to mideleg/hideleg or
   3) each hpmcounter/hpmevent can be assigned to a privilege and LCOFIP should be handled by mode that the counter/event associated.

Thanks,
Steven-Li

gfavor, 2025-01-22:
The Sscofpmf spec (chapter 20 Priv spec) says (with my added bold highlighting):

> This count-overflow-interrupt-request signal is **treated as a standard local interrupt** that corresponds to
> bit 13 in the mip/mie/sip/sie registers. The mip/sip LCOFIP and mie/sie LCOFIE bits are, respectively,
> the interrupt-pending and interrupt-enable bits for this interrupt. ('LCOFI' represents 'Local Count
> Overflow Interrupt'.)
> 
> Generation of a count-overflow-interrupt request by an hpmcounter sets the associated OF bit. When an
> OF bit is set, it eventually, but not necessarily immediately, **sets the LCOFIP bit in the mip/sip registers**.
> The LCOFIP bit is cleared by software before servicing the count overflow interrupt resulting from one
> or more count overflows. The mideleg register controls the delegation of this interrupt to S-mode versus
> M-mode.
>

And, as is true for all interrupts, the standard *ideleg-based interrupt delegation mechanism applies.

gfavor, 2025-02-07:
If I understand your question correctly ... the RISC-V interrupt/exception delegation mechanism applies equally for all bits in the the *ip/*ie/*ideleg CSRs.  So LCOFI can be delegated down to one specific privilege mode and, since all perf cntr overflow interrupts are lumped together into the LCOFI local interrupt, they all get delegated together.

closed 2025-03-31 by Steven-Li-Xiaogang