riscv/riscv-isa-manual #1814: What does "inlined" mean?

kersten1, 2025-01-17:
In the 14.1. Specifying Ordering of Atomic Instructions section of the unpriv, there is a para that says this: 

"LR/SC can be used to construct lock-free data structures. An example using LR/SC to implement a compare-and-swap function is shown in Example 2. If inlined, compare-and-swap functionality need only take four instructions."

Inlined isn't a word that I can find. Bill T pointed out that it is probably in reference to the inline assembler. 

Can someone confirm that? 

If so, Does this rewrite make sense?

"LR/SC can be used to construct lock-free data structures. The previous example shows LR/SC implementing a compare-and-swap function. If this example is embedded inline, the compare-and-swap functionality requires only four instructions."

ved-rivos, 2025-01-17:
"Inlined" refers to code that has been directly inserted into a sequence of instructions rather than being referenced as a separate function. It’s a well-understood term in programming and compiler design, often associated with optimizing performance by avoiding the overhead of function calls. However, if clarity is a concern, "embedded inline" might be a better choice for broader audiences.

aswaterman, 2025-01-17:
IMO, “inline” is a sufficiently common verb in the programming languages/compilers context that this sentence is fine as-is.

kersten1, 2025-01-21:
@aswaterman @ved-rivos 

"inlined" is not a word. I can infer meaning from it based on the common usage of "inline" but it doesn't change the fact that "inlined" is not a word! Additionally, the sentence has an unclear antecedent. You must infer that it is the code from the example that is to appear inline. Plus do all compare-and-swap functionalities that are embedded inline require only four instructions or just this example? 

Rather than make the reader infer meaning, we can make this section more clear with a few edits.

We can, however, just remove the "d" from "inlined". It is a less effective fix, but one I can live with.

aswaterman, 2025-01-21:
It's common practice for technical documents to use terminology that's nonstandard in English but is standard within the technical domain.  In this community, using "inlined" as a past-tense verb or as an adjective is without a doubt accepted: https://scholar.google.com/scholar?hl=en&as_sdt=0%2C5&q=inlined&btnG=

Of course we could rewrite this one sentence, but I'd like to first understand the broader strategy.  This document is chockablock with computer-science jargon.  Rewriting the document to be entirely free of technical jargon would seem to me to be a questionable expenditure that would ultimately produce something that's less readable.

allenjbaum, 2025-01-21:
I'm going to have to agree with Andrew here; the description of something
as "inlined" will be commonly understood  by anyone who can understand the
rest of this document.
It's a software description that even I, as a hardware person, understand.
It refers to a function that is inserted directly into software instead of
being called as a subroutine.
Just wait until you get to "loop unrolling"....

kersten1, 2025-01-21:
Let’s step back a bit. I have absolutely no problem with the word “inline”. It has a common meaning that even I understand! The word “inline” absolute has a place in technical documentation, including the RISC-V documents. In fact, in my first comment, my suggested rewrite includes the word “inline”. My objection is rather to the form “inlined”.  

We are moving towards becoming an ISO standard, which indicates the following spelling references:

8.2 Spelling reference works 

Spelling shall be consistent throughout a document. The following reference works for spelling are suggested: 

• for English, the Shorter Oxford English Dictionary, the Concise Oxford English Dictionary, the Collins Concise English Dictionary, the Webster’s New World College Dictionary or the Chambers Concise Dictionary; 
• for French, Dictionnaire Le Robert, Dictionnaire Larousse and Dictionnaire des difficultés de la langue française (A.V. Thomas, Larousse).

While “inline” appears in all of them, “inlined” does not. 

While I agree that this documentation is full of technical jargon, as it should since it is a technical document, I do think that we can make the language easier by rewriting some technical jargon into common word choices.

aswaterman, 2025-01-21:
I'm not trying to make a big deal out of "inlined".  My only goal is to derive a general rule, since we're sure to run into this issue repeatedly.

I respectfully disagree with your interpretation of that ISO directive.  By my reading, its only requirement is consistency within a document.  The references the directive mentions are only suggestions; there's no requirement that a word show up in one of them.

kersten1, 2025-01-21:
I am not trying to make a big deal either! If the general rule you are trying to derive is that technical jargon is allowed, I'm all for that! For example, "virtualized" isn't found in most dictionaries. But I would never suggest removing it from technical documentation. Which is why I was asking for a definition of "inlined" just to make sure that I was not missing out on a specialized definition. I will try to be more clear in my requests in the future.

I will also respectively point out that, if consistency is our goal, "inlined" appears one time in all of the priv and unpriv PDFs.  Not sure of other docs such as debug.

allenjbaum, 2025-01-21:
A (very) quick websearch results in "inlined" being primarily an adjective
("is inlined"  or "an inlined function"),
overlapping with "inline" as a verb ("to inline a function") and also an
adjective ("an inline function is"
I'm not sure I can come up with a good rule when to use one rather than the
other, except on is defining the concept, and the other describes how it is
used?
Subject vs object maybe?
------------
How to guarantee that a function is inlined in C++
MSVC - Any way to check if function is actually inlined?
Identify whether a function is inlined in the LLVM IR
How do I know if a function has been inlined or not?
An inline function is one for which the compiler copies
How to check if a function is inlined or not?
ADMIN_IS_INLINED function - Determine if data is inlined
Perform better optimizations spanning the code the function is inlined into
that it normally couldn't perform across function boundaries.
... if the function the lambda is passed to is inlined, the return can be
inlined, as well.
------------
What does it mean to inline a function and how does it affect a program?
(IBM.  https://www.ibm.com › support › pages › what-does-it-m...
inline Functions in C and C++
https://kotlinlang.org › ... › Functions › Inline functions
In computing, inline expansion, or inlining, is a manual or compiler
optimization   <-- both forms in one sentence
In the C and C++ programming languages, an inline function is one qualified
with the keyword inline; this serves two purposes.
Inline parameters do not create bindings and are simply inlined.  <-- both
forms in one sentence

jjscheel, 2025-01-22:
A couple observations:

1. The RISC-V ISA manual has been intermittently criticized for its readability and style.  The question posed here goes to both areas.  I hope we are in agreement that it's worth our time and effort to invest in improving the readability and style of our specifications.  I also hope we recognize that while this particular issue seems trivial, there are dozens of issues just like it which will need to be worked to begin to significantly improve readability of our specifications.  Thus, while it may appear we're obsessing over a minor issue, we're actually attempting to build a method for handling other readability items as we move forward.
2. The foundational question (which feels like it's been missed) is what dictionaries we're going to use for guidance ("rules").  ISO suggests some including a couple Oxford variations.  Once we have (at least) one, we should then agree to apply it diligently.  If we're not going to do so, there's no sense picking one.  Yes, there will be exceptions that we can handle, but they should be few (by the definition of _exception_).
3. We all have different skill sets:  some are experts at designing an ISA, others are experts at designing text.  In areas of grammar, we should encourage the grammar experts to propose text improvements (based on the rules, from our agreed upon dictionaries) and let the ISA experts ensure the proposed text does not change the technical meaning.  This will be a collaborative process, but it will not function effectively without order.

If we can agree on these three things, I feel like we can make progress.  If not, I worry whether and how we will improve.

radimkrcmar, 2025-01-22:
> 1. [...] we're actually attempting to build a method for handling other readability items as we move forward.

Well said, I'll try to continue along these lines:

> 2. The foundational question [...] is what dictionaries we're going to use for guidance. [...]

The ISO suggested dictionaries don't have "inline"; they only have different meanings of "inline", which makes them inapplicable.  Similar cases in the future could be more conflicting if we choose dictionaries that do not keep pace with language evolution: here one side claimed "inline" (verb) is not a word while the other side understood it perfectly.

For example following links from [inlined](https://en.wiktionary.org/wiki/inlined) can get a reader to an explanation of [inlining](https://en.wikipedia.org/wiki/Inline_expansion).  LLMs also give a good explanation.
Not that I propose to use either.   I just think that having unnecessarily strict guidelines can result in worse cooperation.
(I want to draw similarities to the letter of law against the spirit of the law.)

> 3. [...] In areas of grammar, we should encourage the grammar experts to propose text improvements [...] and let the ISA experts ensure the proposed text does not change the technical meaning. [...]

As I see it, this more or less happened here.  The  "inline" (verb) has not been understood and the replacement phrase "embedded inline" was rejected because it's not a common expression for the intended meaning.

There were a few tangents, but sharing ideas is not an easy process.

kersten1, 2025-01-23:
Going to try one more time to explain this issue. The word we are discussing is "inlined". The sentence we are looking at is "If inlined, compare-and-swap functionality need only take four instructions." In this case, the word "inlined" is an adjective, describing the example code (as inferred from context - see original issue description for full passage). You can determine this by adding a noun to the sentence. You would not say "If the code inlined" but instead "If the code is inlined." So "code" is the noun, "inlined" is the adjective that is describing the code and "is" is the linking verb, connecting the noun to the adjective.

So let's substitute another adjective, for example "blue". Because blue doesn't have any meaning really in the computer world, let's use a car as the noun. So "If blued, the car moves to the left." Okay so what does "blued" mean in this case? I can infer meaning from the base word "blue" and assume that for some specific reason, blue cars go left. But, this sentence occurs in highly technical information so I need to make sure that "blued" does not have any specialized meaning (and hence, I created the issue). However, it turns out that "blued" means the same as "blue". To bring it full circle, "inlined" has been determined to have the same meaning "inline".

So now we have what is basically a typo or new technical jargon. We can either fix this single use word by removing the "d" (and preferably adding a noun and linking verb) or we can set up a entry in our glossary and create a new word, complete with definition, as you should always do when introducing technical jargon, even sometimes with jargon that is commonly understood. Again, this new word in "inlined", which is a new word and not "inline", which is not.

We can have a whole discussion about the verbification of nouns and adjectives, but it really isn't a best practice in technical documentation.

radimkrcmar, 2025-01-23:
"[inlined](https://en.wiktionary.org/wiki/inlined)" in that sentence is a past participle of the verb "inline". 

It is not productive to discuss solutions without agreeing on the issue...  We want to make a precedent with a good explanation of the underlying cause.  So far, I have no idea if the problem is one or more of the following:
1. participle
2. past tense
3. same spelling but a different meaning
4. domain-specific word

kersten1, 2025-01-23:
I don't think we want to use Wiktionary as a technical reference. For example, "[blued](https://en.wiktionary.org/wiki/blued)" is also defined in WIktionary. 

If we are trying to stick to the words found in the approved dictionaries, then "inline" is generally defined as an adjective.

If the bad word in my rewrite is "embedded", then I have no problem removing it and changing to "If this example is inline". I'll even go for "If inline". Neither of these suggestions change the original meaning of the sentence. We generally do not use past tense in technical writing as a best practice anyway.

radimkrcmar, 2025-01-23:
Yes, that is the root of this issue.  Dictionaries don't have the definition of "inline" that is used in this manual.
That alone is not a reason for expulsion for me, though.  We should strive to pick the best word.

For example [Ada 22 (ISO/IEC 8652:2023)](http://ada-auth.org/standards/ada22.html) uses inlined:
> The implementation shall document the overhead associated with calling machine-code or intrinsic
subprograms, as compared to a fully-inlined call, and to a regular out-of-line call.

Blindly copying others isn't good practice either, so if we look for alternatives, then "If expanded inline," is the most common one (I would never use it as it feels baroque, but it should be understandable).
The tense still needs fixing, but I feel like we can move from here.

jjscheel, 2025-01-24:
Before I add a few new thoughts, thanks to all who've been pushing this discussion forward in a productive manner.  It may not feel like it, but we're making progress! As a good friend of mine said to me once, "It's not the result, it's the process which matters."  It's funny how many situations that applies to in life...

Returning to the problem at hand, I believe we need to agree upon our "rules", i.e one or more dictionaries.  Any one we use should be freely available **on-line** so  that we can use them and link to them to answer questions like "is _inlined_ a word?" and "what part of speech is _inlined_?"  Having these would allow us to use the same "rules" for this and similar issues.

If we agree, I propose the @kersten1 and the Doc SIG take the AI to provide the "definitive" list and ensure we put it in our specification writer's guide ([_Authoring and Editing RISC-V Specifications_](https://drive.google.com/file/d/1Xbq436wf3HYC7qwlaw7U0mj19Vf0Myn4/view)).

Secondly, I'd propose we should take this time clearly articulated process, in the writer's guide again, for handling our exceptions.  Specifically, I'm interested in seeing the Where and the How?  Although I think the Where is most simple (our Glossar), I'll be intrigued to see the How.  For example, is the Doc SIG going to review and approve all proposed new words or are you going to let me decide (Mom would be so proud if you do).  ;-)

If we do these two things, we can all apply the rules and if we still feel we need to create an exception, we know what's needed.  Then, we can complete our fun with _inlined_.

kasanovic, 2025-01-29:
As @radimkrcmar says, inlined is being used as past participle of verb "inline".  Consider another English verb "delay" in this following excerpt from a fictional train conductor instruction manual:
"If the train is on time, please close doors no sooner than five minutes before departure.  If delayed, close no sooner than three minutes before departure."

radimkrcmar, 2025-01-30:
> Returning to the problem at hand, I believe we need to agree upon our "rules", i.e one or more dictionaries. Any one we use should be freely available **on-line** so that we can use them and link to [...]

Sounds good.
 
> I'll be intrigued to see the How. For example, is the Doc SIG going to review and approve all proposed new words or are you going to let me decide (Mom would be so proud if you do). ;-)

I believe the natural process is
1. A Doc SIG member recognizes a word that is not in the chosen dictionaries.
3. The lucky member sends a pull request.  (For the RISC-V dictionary or the obscure manual.) 
4. A hearty discussion happens, and domain experts might be dragged in to clarify.
5. An agreement is reached, or no-one has the willpower to continue bike-shedding.
6. Whoever has the merge rights decides.

aswaterman, 2025-12-09:
Closing this issue, as happily we've resolved most of these diction topics.

closed 2025-12-09 by aswaterman