riscv/riscv-isa-manual #1817: No way to distinguish exceptions coming from different memory protection levels to M-mode handler.

xxkent, 2025-01-20:
Let's assume that the system has 4 units of memory protection: M level PMP(ePMP) + HS level MPU(MMU) for virtual machines + VS level MPU(MMU) + S level MPU(MMU).
The "The RISC-V Instruction Set Manual", version 20241101, table 43 (Machine and supervisor cause register (mcause and scause) values when the hypervisor extension is implemented) defines the following exception values:
12 - Instruction page fault
13 - Load page fault
15 - Store/AMO page fault
20 - Instruction guest-page fault
21 - Load guest-page fault
23 - Store/AMO guest-page fault

The question is about handling exceptions at the M level (interrupts are not delegated). 

If exceptions come to the handler at the VS level, then just numbers 12,13,15 are used. 

Exceptions coming to the handler at the HS(S) level can be caused by the HS, S or VS level protection, and here we will use numbers 12,13,15,20,21,23 values and hstatush.spv(V) to distinguish whether an exception has come from VS or S level memory protection unit. If from HS level unit: 20,21,23 and V=1. If from VS level: 12,13,15 and V=1. If from S level: 12,13,15 and V=0.

But if everything goes to the M level handler, then I believe we currently have no way to distinguish who raised the exception because in addition to what is described above for the HS(S) level, we are given exceptions coming from the M level PMP(ePMP). We can still distinguish exceptions coming from the HS protection level because they have dedicated interrupt numbers 20,21,23. But VS level MPU raises the same exception values as M level PMP: 12,13,15. The same for S level MPU and M level PMP: 12, 13, 15.

How can we distinguish in this case all exceptions coming to M-mode handler from all memory protection units?

allenjbaum, 2025-01-20:
can't you look at status.mpp and .mpv to determine which mode they came
from?

xxkent, 2025-01-20:
The question here is rather the following. Let's assume that the code was executed at level S or U and raised an exception. Yes, we know in M-mode handler that the exception came from level S or U. But how can we understand whether it was caused by the MPU of level S or the protection was triggered by PMP unit, they have the same exception numbers.

At the HS level and above, things are simpler. Exceptions caused by M-level protection do not get there(they remain at level M). And if we find ourselves in such a handler, we can be sure that the M-level did not cause an exception. If I understand this correctly.

allenjbaum, 2025-01-20:
PMP triggers  illegal access faults, not page faults;
MPU (which is either MMU or SPMP) trigger page faults.

xxkent, 2025-01-21:
Confused this with SPMP. Thanks for clarifying!

closed 2025-01-21 by xxkent