riscv/riscv-isa-manual #1818: Smepmp extension chapter needs a rewrite to make it normative

radimkrcmar, 2025-01-21:
The text that made it into this repository is not a specification, but a proposal, and needs a thorough rewrite.

To list some issues:
> 6.2. Proposal

> The reset value of mseccfg is implementation-specific, otherwise if backwards compatibility is a requirement it should reset to zero on hard reset.

> 6.4. Rationale

A lot of the Smepmp chapter should be straight deleted.

aswaterman, 2025-01-21:
Yes, overall, the effort to merge all the specs together has been structural rather than semantic.  Each spec's authors should be working with @wmat to close that gap.

gfavor, 2025-01-21:
Given the more substantial than typical rewrite that is required, this need
should be raised to the Security HC chair (Andrew Dellow) to ensure that
someone commits to doing the work and doing it sometime soon.

Cc'ing @Andrew Dellow ***@***.***>

Greg

kasanovic, 2025-01-27:
This comment can be applied to many sections of the manual.

That said, we should also be keeping an archive of the proposals which often include rationale that helps explain design details, and which are useful for later development.  But I agree this should be normally not in the reference manual, except where it helps clarify the specification itself.

radimkrcmar, 2025-01-27:
Rationale in the spec is welcome, as long as it is actually rationale and is clearly marked as non-normative.

If you look at Rationale section, it has sentences

> Since a CSR for security and / or global PMP behavior settings is not available with the current
spec, we needed to define a new one. [...]

that only made sense while the proposal was outside the manual.

Smepmp has issues in normative text as well.  A lot of definitions around mseccfg are outdated, because the CSR is already defined and used outside of this extension.  For example, the sentence I already quoted

> The reset value of mseccfg is implementation-specific, otherwise if backwards compatibility is a requirement it should reset to zero on hard reset.

is barely readable even with the usual amount of goodwill needed for the manual.  (The manual should say if the value is implementation specific or reset to zero.  Based on the reset section, I think it is neither.)

Timmmm, 2025-02-25:
Some other bits that need to be reworked:

> However, there are no such mechanisms available on Machine mode in the current (v1.11) Privileged Spec. ...

> Machine Security Configuration (mseccfg) is a new RW Machine mode CSR

> All mseccfg fields defined on this proposal are WARL

> The reset value of mseccfg is implementation-specific, otherwise if backwards compatibility is a requirement it should reset to zero on hard reset.

(The previous two should be moved to where `mseccfg` is defined.)

> On mseccfg we introduce a field

(There are a few more "we introduce a field"s.)

> Adding a rule

(Not related to it being a proposal, but "Adding a rule" is an odd way to say "Configuring a PMP entry".)

> Visual representation of the proposal

> It is expected that BootROM

(Probably should be "the firmware" or similar.)

> Since a CSR for security and / or global PMP behavior settings is not available with the current spec, we needed to define a new one. This new CSR will allow us to ....

----

Also

> Machine-Mode Allowlist Policy (mseccfg.MMWP)

Based on the acronym I think this is supposed to be Machine-Mode Whitelist Policy? If you think the whitelist is racist then, how about `Machine-Mode Waiver Policy`? I think that's probably a better name anyway.

I actually really like the Rationale section. It would be nice if more of the spec had that.

mickflemm, 2025-03-03:
Updated both the text and the image, let me know if that works:
https://github.com/riscv/riscv-isa-manual/pull/1879