riscv/riscv-isa-manual #1826: Priv & Unpriv ISA manuals use the wrong heading level for colophons

james-ball-qualcomm, 2025-01-26:
For a new backend generator for the riscv-unified-db (a Certification Test Plan generator for the CSC), I've found a problem with how the adoc for the manuals is written in a couple of tiny ways. 

I'm using the "book" and "part" features to add the priv and unpriv ISA manuals as "parts" to a riscv-unified-db adoc file which is the first part. See the screenshot below of my generated PDF that has all 3 parts in one PDF after I made the change described below.

At first, the ISA manuals weren't being recognized as "parts" by asciidoctor and I got the following error message:
    asciidoctor: ERROR: ext/riscv-isa-manual/src/colophon.adoc: line 2: invalid part, must have at least one section (e.g., chapter, appendix, etc.)
    asciidoctor: ERROR: ext/riscv-isa-manual/src/priv-preface.adoc: line 2: invalid part, must have at least one section (e.g., chapter, appendix, etc.)

Both adoc files use the "colonphon" feature of adoc and were both using a level 0 heading (one equals sign) which matches what the asciidoctor documentation says you should do when using a book doctype (which the ISA manuals already do). See https://docs.asciidoctor.org/asciidoc/latest/sections/colophon/.

However, following the asciidoctor documentation created the ERRORs above and also prevented asciidoctor from recognizing the existing level 0 headings that are near the start of each ISA manual. The asciidoctor doc says to use a level 1 heading (two equals sign) if not using book doctype. So, I tried an experiment and changed the colonphons to use a level 1 heading and yo and behold, the error message went away and asciidoctor recognized the ISA manuals as parts of a book. Yay!

I tried building the ISA manuals in the riscv-isa-manual workspace the normal way (as separate PDF files) and it looks like it works fine to me and I didn't see any error messages in the asciidoctor output. So, I'd like you to consider accepting a PR from my fork of the repository into the main.

[image]

wmat, 2025-01-27:
I tested as well and it doesn't seem to break any of the builds.  Did you want to go ahead and issue a PR?

james-ball-qualcomm, 2025-01-28:
> WARNING: This email originated from outside of Qualcomm. Please be wary of any links or attachments, and do not enable macros.

Sure

+james+

james-ball-qualcomm, 2025-02-03:
There is one other change I had to make to allow proper linking to adoc anchors in the Priv and Unpriv ISA manuals. Currently, they both have the same adoc anchor at the top of their adoc files ([[risc-v-isa]] in riscv_unprivileged.adoc and riscv_privileged.adoc). I don't see any links to these anchors so propose we change them to be unique and add default reftext. 

So, my proposal is to use:

- [[manual:priv,RISC-V ISA Manual Volume II: Privileged Architecture]]
- [[manual:unpriv,RISC-V ISA Manual Volume I: Unprivileged Architecture]]

wmat, 2025-02-03:
Fine with me so long as nothing breaks in the TOC or on the Title page.

james-ball-qualcomm, 2025-10-01:
This has been merged into main so can close it.

closed 2025-10-01 by james-ball-qualcomm