riscv/riscv-isa-manual #1828: Question about PMA

oolegoon, 2025-01-27:
Could you please clarify some PMA statements which is unclear for me?

1) Spec states:
_Coherent main memory regions always have either the RVWMO or RVTSO memory model. Incoherent main
memory regions have an implementation-defined memory model.
Accesses by one hart to an I/O region are observable not only by other harts and bus mastering devices but
also by the targeted I/O devices, and I/O regions may be accessed with either relaxed or strong ordering._

Can non-cacheable main memory have strongly-ordered channel 0 PMA? Or main memory can have only RVWMO or RVTSO memory model in case of coherency and implementation-defined memory model (which can be similar to strongly ordered channel 0) in case of incoherency?

2) Spec states:
_Channel 1 is used to provide global strong ordering across all I/O regions. Any accesses by a hart to any I/O
region associated with channel 1 can only be observed to have occurred in program order by all other harts
and I/O devices, including relative to accesses made by that hart to relaxed I/O regions or strongly ordered
I/O regions with different channel numbers. In other words, any access to a region in channel 1 is
equivalent to executing a fence io,io instruction before and after the instruction._

What does channel 1 mean? What is meant by program ordering across all harts? Isn't the term program order applicable only to one hart? In a multi-core system where hart=core channel 1 orders IO requests across all cores?

gfavor, 2025-01-28:
The key to properly reading and interpreting the spec is to recognize that "main memory" and "I/O regions are two distinct types of regions with different properties.

So I/O channels (0,1, 2, 3, etc.) only apply to I/O regions and are irrelevant to main memory regions.  Which is why main memory regions always have one of the two specified memory models.

Performing a channel 1 access, as spec'ed, is equivalent to executing a fence io,io before and after a channel access.  This forces an ordering of this access wrt ALL preceding I/O accesses and wrt ALL following I/O accesses.  Whereas without this fencing behavior, this access would only be ordered wrt all other accesses in this same channel (i.e. channel 1).

oolegoon, 2025-01-28:
Thank you for your response.

I think I got it.

Let's say we have 4 channels: 1channel0, 2channel0, channel1, channel2

In 1channel0 requests: 1, 2
In 2channel0 requests: 3, 4
In channel1 requests: 5, 6
In channel2 requests: 7, 8

In the program order of one chart they have the following positions 1, 2, 3, 4, 5, 6, 7, 8

It turns out that 1, 2, 3, 4 can be mixed with each other, only 2 cannot overtake 1, and 4 cannot overtake 3.
5, 6 are executed strictly one after another, after 4 and before 7.
Is it right?

gfavor, 2025-01-28:
Yes in that requests within channels that aren't channel 1 are ordered only wrt other requests in the same channel.  While requests in channel 1 are ordered wrt all requests across channels.

gfavor, 2025-01-29:
Closing since I believe the question has been resolved.

closed 2025-01-29 by gfavor