riscv/riscv-isa-manual #1829: PMP write-only region

mateoconlechuga, 2025-01-27:
I would like to have a write-only region in all modes (M/S/U) and it doesn't appear that the specifications support this ability. Is there any chance that this could be further clarified? This seems to be a rather crucial missing aspect - write-only memory is important for other masters in the system.

aswaterman, 2025-01-27:
You are right that the PMP standard doesn't support write-only regions.

Other masters (i.e. things that aren't harts) are not subject to PMP, anyway, but they could be subject to other mechanisms that effect write-only permissions.

gfavor, 2025-01-27:
I can't check at the moment, but does ePMP support write-only regions?

Greg

aswaterman, 2025-01-27:
I believe you're thinking of the fact that ePMP appropriated the bit patterns that would have correspond to write-only regions for new region types, but these new region types are not write-only.

mateoconlechuga, 2025-01-27:
Yeah, a write-only region from the perspective of the CPU would be especially helpful, in order to protect the data being read/replayed by software if it only ends up being consumed by hardware (thinking of security/encryption of data)

ved-rivos, 2025-01-27:
Could you explain the threat model and use case a bit more. For instance the producing agent can always write what was written before to replay the write and it already knows what was written without needing to read it back. So is this for temporal isolation where some lower integrity agent may have access to that memory after a higher integrity agent has produced some data ?

allenjbaum, 2025-01-28:
There are currently no standards that support write-only regions formPMP,
ePMP, or sPMP.
But, a PMA can certainly allow a write only region, I believe.
The problem is that while the PMA  schema is defined, there is
no standard way to describe them that can be used between DUTs and
simulators.

mateoconlechuga, 2025-01-28:
@ved-rivos Here's an example: the CPU writes an encrypted piece of data into memory, which is then read directly by hardware without CPU intervention, such as sending the encrypted packet over a network. An attacker can craft a program to read the contents of the encrypted memory, and replay the packet at any time. This is primarily for M-mode operation since the MMU would generally provide isolation between threads as well

ved-rivos, 2025-01-29:
> An attacker can craft a program to read the contents of the encrypted memory, and replay the packet at any time. This is primarily for M-mode operation since the MMU would generally provide isolation between threads as well

Okay the threat model is : So the threat model is that M-mode crafts an encrypted packet and puts it in a region which is accessible to **lower privileges** and then the lower privilege reads the encrypted packet and replays it. If the region was only accessible to M-mode then such a replay would need a M-mode compromise. Is this shared memory region with the network controller supposed used as a fifo? How do the lower privilege and M-mode sharing this fifo agree on use of metadata for the FIFO such as head and tail pointers.

mateoconlechuga, 2025-01-29:
I'm not sure if head and tail pointers make any difference, it's a hardware peripheral so M-mode has access to all registers. Yes, it requires an M-mode exploit, but then the question is what is the purpose of the PMP in the first place if it can't be used to help protect against M-mode exploits.

ved-rivos, 2025-01-29:
> I'm not sure if head and tail pointers make any difference

Head and tail pointers matter for the replay threat model. Consider a fifo that has wrapped around once. Just advancing the tail by 1 will put the oldest stale packet back in the fifo and so one can replay that packet without needing to read it.

mateoconlechuga, 2025-01-29:
Yes, software changing the pointers could effectively do the same thing as long as the hardware wasn't marking the packets as complete or clearing them after use. It also limits the attack to just what is in the ring buffer.

M-mode software (at least in our case) never has direct access to the encryption keys, encryption is always performed by external hardware modules, the CPU is just responsible for moving the data where it needs to go in memory for the hardware to access it.

I agree that this is a use case that is rather niche, we were just hoping to be able to use the PMP to restrict read access to a section of memory rather than creating a dedicated hardware bus firewall or similar.

mateoconlechuga, 2025-01-29:
This page kind of describes the use cases: https://en.m.wikipedia.org/wiki/Write-only_memory_(engineering)

mateoconlechuga, 2025-01-29:
I would also probably clarify that this may not always require an actual M-mode exploit - the main goal to having write-only memory is so programs cannot read what other programs wrote to hardware. Whether or not that requires an exploit is kind of moot.

mateoconlechuga, 2025-01-29:
This table doesn't really make any sense to me: 

[image]

It would be so much cleaner to just add an extra bit to the PMP register rather than hijacking the write bit. :(

ved-rivos, 2025-01-29:
The network use case is still unclear to me - packets transmitted to the network needing replay protection while in the memory FIFO but not on the network, packets received received from the network do not needing confidentiality, integrity, or replay protection, two programs both running in machine mode but mutually distrusting each other. However, I don't need to know specifics.

mateoconlechuga, 2025-01-30:
@allenjbaum I have no clue what fifo or hardware you guys are even talking about... I fear that there's some miscommunication here. I'll try to clear things up - please don't make any assumptions because someone started talking about head and tail pointers and I have no idea what that has to do with anything.

1. Sensitive data is written to memory (not a device) 
2. This sensitive data should not be allowed to be read *at all* from the CPU after it is written

If RISC-V can't do this, then I'm going to have to modify the memory bus to protect the region of memory (decode error, slave error, or similar). My goal was to use the PMP to do this, but it sounds like it can't, so here we are.

aswaterman, 2025-01-30:
@mateoconlechuga Agreed with your assessment of the status quo.

aswaterman, 2025-01-30:
/

closed 2025-01-30 by aswaterman

mateoconlechuga, 2025-01-30:
@aswaterman, sure but I would just like to say that the entire PMP specification is kinda horrible. This is a valid use case, and the fact that it doesn't support it doesn't give me any confidence in the insanity that the PMP has become.

gfavor, 2025-01-30:
The PMP spec was ratified a long time ago.  Then a TG a few years ago
developed the Enhanced PMP spec, and for better or worse apparently that
group and the spec didn't see fit to address this case as part of all the
enhancements.

But there is always room for new extensions to improve upon what already
exists.  Pursuing that would start with raising this issue with the
Security HC.

Greg

aswaterman, 2025-01-30:
Please keep feedback constructive.  This is an example of how not to do that.

allenjbaum, 2025-01-31:
I’ll concur with what Andrew said, in every respect.
This actually does violate standards that are spelled out in the videos
that are played before every meeting.

This, and ePMP, have been around for years, and this is the first time that
I am aware of that this issue has come up, so something that appears to be
obvious to you is clearly not obvious to everyone ( or anyone) else.

Suggesting a fix for this (with the rationale- it took several rounds of
discussion and explanation before it was clear what was going on) might be
much more
 productive than railing about how horrible you think the current state of
affairs is.