riscv/riscv-isa-manual #1830: Change "implementor" to "implementers"; fix a couple typos

kbroch-rivosinc, 2025-01-28:
Mostly did this as way to find terms not in the vocabulary of `codespell` (that we might want to include when running `vale` @kersten1 ).  But did find some typos and ["old" English](https://www.oxfordreference.com/display/10.1093/acref/9780197599020.001.0001/acref-9780197599020-e-4589?rskey=9Pa8mp&result=1). 

Here's results of initial run of `codespell` and getting counts of things it thinks are spelling errors:

```
~/rvi/repos/riscv/riscv-isa-manual on main                                                                                                   at 11:27:39 AM
❯ codespell src | cut -d ' ' -f2  | sort | uniq -c  | sort -nr
 282 hart
 219 edn
  42 Hart
  41 sie
  25 STIP
  22 vill
  22 SIE
  10 SXL
  10 FLE
   9 covert
   6 vor
   5 implementor
   5 bund
   5 SHS
   4 useed
   3 implementors
   2 rouge
   2 USEED
   2 INH
   2 Covert
   2 Colum
   2 BU
   2 ADDD
   1 unstalls
   1 spcifiers
   1 ois
   1 inbetween
   1 Synopsys
   1 Specifiations
   1 Jaques
   1 Implementors
   1 Commun
```

Most of the above are words/term, so made exceptions for those, fixed the others and now it runs clean:

```
~/rvi/repos/riscv/riscv-isa-manual on dev/kbroch/codespell-corrections:main wip ⇡1                                                           at 11:31:03 AM
❯ codespell -L edn,hart,Hart,sie,STIP,vill,SIE,SXL,FLE,rouge,bund,covert,useed,vor,SHS,Synopsys,ADDD,BU,unstalls,INH,Colum,inbetween,Commun,Jaques,ois  src ~/rvi/repos/riscv/riscv-isa-manual on dev/kbroch/codespell-corrections:main wip ⇡1                                                           at 11:31:24 AM
❯
```

ved-rivos, 2025-01-28:
Some of these waivers may need to be fixed:

- rouge -> rogue
- Synopsys -> synopsis
- Commun -> Common
- bund -> bound
- Colum -> Column

kbroch-rivosinc, 2025-01-28:
> Some of these waivers may need to be fixed:
> 
> * rouge -> rogue
> * Synopsys -> synopsis
> * Commun -> Common
> * bund -> bound
> * Colum -> Column

@ved-rivos : thanks for taking a look.  I'll make change noted below:

generated clickable links from your suggestions:
```
❯ codespell -L edn,hart,Hart,sie,STIP,vill,SIE,SXL,FLE,covert,useed,vor,SHS,ADDD,BU,unstalls,INH,inbetween,Jaques,ois  src | gsed -e 's/\:/?plain=1#L/' | xargs -I {} echo "https://github.com/riscv/riscv-isa-manual/blob/main/{}"
```

https://github.com/riscv/riscv-isa-manual/blob/main/src/history.adoc?plain=1#L76: Synopsys ==> Synopsis
https://github.com/riscv/riscv-isa-manual/blob/main/src/riscv-privileged.adoc?plain=1#L38: rouge ==> rogue
https://github.com/riscv/riscv-isa-manual/blob/main/src/riscv-unprivileged.adoc?plain=1#L35: rouge ==> rogue
https://github.com/riscv/riscv-isa-manual/blob/main/src/scalar-crypto.adoc?plain=1#L5344: Colum ==> Column
https://github.com/riscv/riscv-isa-manual/blob/main/src/scalar-crypto.adoc?plain=1#L5350: Colum ==> Column
https://github.com/riscv/riscv-isa-manual/blob/main/src/resources/riscv-spec.bib?plain=1#L933: bund ==> bind, bound
https://github.com/riscv/riscv-isa-manual/blob/main/src/resources/riscv-spec.bib?plain=1#L1090: Commun ==> Commune, Common
https://github.com/riscv/riscv-isa-manual/blob/main/src/resources/riscv-spec.bib?plain=1#L1309: bund ==> bind, bound
https://github.com/riscv/riscv-isa-manual/blob/main/src/resources/riscv-spec.bib?plain=1#L1320: bund ==> bind, bound
https://github.com/riscv/riscv-isa-manual/blob/main/src/resources/riscv-spec.bib?plain=1#L1402: bund ==> bind, bound
https://github.com/riscv/riscv-isa-manual/blob/main/src/resources/riscv-spec.bib?plain=1#L1592: bund ==> bind, bound

I think only the `Colum -> Column` end up being a problem.  I probably should have skipped the *.bib page completely.

ved-rivos, 2025-01-28:
> I think only the Colum -> Column end up being a problem.

@kbroch-rivosinc - agree!

kbroch-rivosinc, 2025-02-28:
resolved with https://github.com/riscv/riscv-isa-manual/pull/1831

closed 2025-02-28 by kbroch-rivosinc

kbroch-rivosinc, 2025-03-13:
Just reran codespell:

```
kbroch@kbroch-mac.local:~/rvi/repos/riscv/riscv-isa-manual on  main [!]
❯ codespell -L edn,hart,Hart,sie,STIP,vill,SIE,SXL,FLE,rouge,bund,covert,useed,vor,SHS,Synopsys,ADDD,BU,unstalls,INH,Colum,inbetween,Commun,Jaques,ois  src
src/colophon.adoc:5: messsage ==> message
src/priv-preface.adoc:5: messsage ==> message
src/scalar-crypto.adoc:4516: re-used ==> reused

```