riscv/riscv-isa-manual #1836: Svadu with Svnapot

jiahzhang, 2025-01-29:
Svnapot says

```
If the encoding in pte is valid according to Table 26, then instead of returning the original value of pte, implicit reads of a NAPOT PTE return a copy of pte in which pte.ppn[i][pte.napot_bits-1:0] is replaced by vpn[i][pte.napot_bits-1:0]. If the encoding in pte is reserved according to Table 26, then a page-fault exception must be raised.

Implicit reads of NAPOT page table entries may create address-translation cache entries mapping a + j×PTESIZE to a copy of pte in which pte.ppn[i][pte.napot_bits-1:0] is replaced by vpn[i][pte.napot_bits-1:0], for any or all j such that j >> napot_bits = vpn[i] >> napot_bits, all for the address space identified in satp as loaded by step 1.
```

Given this, is it allowed by the spec to update a PTE with the replaced ppn/vpn bits when hardware A/D bit updates apply? Or should it maintain the original pte.ppn[i][pte.napot_bits-1:0] value?

ved-rivos, 2025-01-29:
Yes, it is allowed to set A and/or D bit in the visited PTE and cache a NAPOT translation. Please see this note:
> If an implementation chooses to use a NAPOT PTE (or cached version thereof), it might not consult the PTE directly specified by the algorithm in Virtual Address Translation Process at all. Therefore, the D and A bits may not be identical across all mappings of the same address range even in typical use cases The operating system must query all NAPOT aliases of a page to determine whether that page has been accessed and/or is dirty.

jiahzhang, 2025-01-29:
Sorry, perhaps my question was not clear. I was more referring to how the ppn/vpn replaced bits apply. From the spec, the implicit read would return a PTE with ppn[i][pte.napot_bits-1:0]  replaced with vpn[i][pte.napot_bits-1:0]. When the atomic update occurs, should it preserve the original ppn[i][pte.napot_bits-1:0] or use the new ppn[i][pte.napot_bits-1:0]?

ved-rivos, 2025-01-29:
> When the atomic update occurs, should it preserve the original ppn[i][pte.napot_bits-1:0] in the PTE or also update the PTE with the new ppn[i][pte.napot_bits-1:0]?

The atomic update must only update the A and/or D bits. It must not modify other contents of the PTE.

closed 2025-01-29 by jiahzhang