riscv/riscv-isa-manual #1837: Are misa.S and misa.U writeable?

Timmmm, 2025-01-30:
The spec isn't really clear but are misa.S and misa.U read-only? All it says is:

> The "U" and "S" bits will be set if there is support for user and supervisor modes respectively.

Is it legal to make a chip where you can disable User or Supervisor mode by writing these bits? (I hope not!)

radimkrcmar, 2025-01-30:
> The spec says that the misa bits are WARL and doesn't condition it further. The implication is that the bits are writable (but implementations might not choose to make them so).
> 
> I'd like to understand why y'all think this is ambiguous. To be clear, I'm not asking for suggestions for additional text; I'd like a clear explanation of why the current text (i.e. WARL + no additional constraints) is ambiguous.

There are additional constraints:

> The "U" and "S" bits will be set if there is support for user and supervisor modes respectively.
> The "X" bit will be set if there are any non-standard extensions.

That are ambiguously worded and only used for these three bits, which makes me wonder if they somehow behave differently.  For me, the most reasonable explanation why this text is normative is that they are read-only-1. 

> > For example, the H extension doesn't depend on S
> 
> This is false. The H spec says it depends on either Sv32 or Sv39, both of which require S.

Right, sorry, I didn't verify that transitive dependency.  (Without a proof, it looks like there could be a wiggle room there.)

aswaterman, 2025-01-30:
> That are ambiguously worded and only used for these three bits
> The "U" and "S" bits will be set if there is support for user and supervisor modes respectively.

I concede the point.  I believe the intent was to describe the reset values of these bits, but one could interpret this sentence to indicate that U and S are hardwired.  The right fix is to reword the sentence to clearly express the originally intended meaning.

Timmmm, 2025-01-30:
Yeah basically what @radimkrcmar said. The "will be set" sentence suggests the bits will *always* be set if the modes are supported, and nothing in the spec suggests that user/supervisor modes *can* be disabled dynamically. It's kind of surprising that they can, so I would expect some explicit text to that effect. At least something like this:

> The "U" and "S" bits will be set if there is support for user and supervisor modes respectively. These bits are optionally writable and can be written with 0 to disable user or supervisor mode. They form a WARL field where U=0, S=1 is illegal.

Does that sound ok?

radimkrcmar, 2025-01-30:
I think that the reset chapter is sufficient when combined with the misa chapter:

> The misa register is reset to enable the maximal set of supported extensions, as described in Section 3.1.1.

> The Extensions field encodes the presence of the standard extensions, with a single bit per letter of
the alphabet (bit 0 encodes presence of extension "A" , bit 1 encodes presence of extension "B",
through to bit 25 which encodes "Z"). 

In those two quotes you can read the difference between "present" and "supported", where "supported" is most likely a synonym to "implemented".  (Should be fixed as the use of synonyms is not allowed in ISO standards.)

The only thing that warrants clarifying what "Supervisor/User mode implemented" means.  (Quoting the table.)
With the non-read-only-1 interpretation, it should read "Supervisor/User mode extension".

Btw. I tried again to prove that misa.H depends on misa.S, but requiring support for SvXX doesn't seem to imply that.
SvXX can be supported without some stuff that is required for full S mode.
(On a tangent, the dependency of H on SvXX could theoretically be lifted, because hgatp can easily be read-only 0.)

aswaterman, 2025-01-30:
> Btw. I tried again to prove that misa.H depends on misa.S, but requiring support for SvXX doesn't seem to imply that.

Look, anyone who's trying to say that Sv32 doesn't require S is obviously just being pedantic.  But if you want to pursue that line of reasoning, please make a separate thread.

aswaterman, 2025-01-30:
> The "will be set" sentence suggests the bits will _always_ be set if the modes are supported, and nothing in the spec suggests that user/supervisor modes _can_ be disabled dynamically. It's kind of [...]

I agree the current wording is problematic, but I'll convene the Priv IC and/or the ARC to address the matter.  No need to submit a PR in the meantime.

aswaterman, 2025-02-06:
The ARC discussed this topic and concluded that the long-time intent has been that these bits are writable, but that the spec could've stated the point more artfully.  The resolution is to rewrite the paragraph to express the intent directly.  This approach is maximally inclusive, since implementers who read the paragraph in the intended way (writable) and those who read it in the more parsimonious way (hardwired) won't have run afoul of the spec.

closed 2025-02-06 by aswaterman

jhauser-us, 2025-02-12:
> The ARC discussed this topic and concluded that the long-time intent has been that these bits are writable

To avoid any confusion, the S and U bits in `misa` *may be* writable but need not be.  It is valid for them to be read-only, of course.  (Or for one to be read-only and the other writable.)

allenjbaum, 2025-02-13:
Or the more complicated case when U is writable, and S is only writable
when U=1 (and is forced to 0 when U=0)

Timmmm, 2025-02-13:
What's the use case for having them writeable? I get that it's *possible* but I can't think of a case where it would actually make sense.

If M mode doesn't want to use S or U mode it can just... not so that, surely?

Making it allowed obviously has costs for software and verification so I feel like there should be better motivation than "because we can". Is there?

allenjbaum, 2025-02-13:
I totally agree, but that's the way the spec(s) have been written.
It is extremely rare for a stupid configuration to be disallowed (and don't
get me started....)

PR #1845 (2025-02-06): Clarify that misa.S and misa.U are optionally writable
This has long been the intent, but the spec didn't make this especially clear.

The proposed resolution is to clarify that these bits are writable, which is maximally inclusive of both classes of implementations: those that choose to hardwire the bits and those that choose to make them writable.

Resolves #1837