riscv/riscv-isa-manual #1850: question for smepmp extension

xt1104, 2025-02-11:
Hi all:
         Is there an official coding guide for enable smepmp extention ？

         For example, when configuring the mseccfg registers, is it necessary to modify the pmp registers simultaneously? Is there a code constraint such that：a sequential requirement for the modification of the mseccfg registers and the pmp registers?

         IF not necessary，after setting the RLB\MMWP\MML field of the mseccfg register, is it necessary to execute sfence. vma  instruction？

gfavor, 2025-02-11:
This software-usage question would be best directed to the Security HC (since the ePMP TG is no longer) at security@lists.riscv.org.

gfavor, 2025-02-11:
P.S. Members of the Security HC were also involved with the development of the ePMP extensions and would be best to comment about coding/usage guidelines.  So closing this issue on the ISA manual.

closed 2025-02-11 by gfavor