riscv/riscv-isa-manual #1851: No Makefile target to create Docbook XML output

james-ball-qualcomm, 2025-02-11:
In today's Documentation SIG meeting I took the action to see if the Makefile supports XML generation. I don't see one I'll add it in a PR.

Note: This is required to be able to work towards creating a tool to extract contents of anchors for the CSC test plan creation.

@Timmmm, hopefully you find some time to create a Makefile target to extract the anchor names and anchor contents for anchors following the naming scheme agreed upon in #1397. In summary, any anchor name that starts with "manual:".

wmat, 2025-02-11:
My mistake, I meant EPUB.

closed 2025-02-11 by wmat

Timmmm, 2025-02-11:
We don't need this for extracting the tags. I did try the docbook XML output but it wasn't very useful because the `manual:foo` IDs end up in XML `id=`s and it's not actually valid to have `:` in XML IDs (it's fine in HTML though). So in the end I wrote a custom Asciidoctor backend to output the tags/contents directly as JSON instead.

See my comment [here](https://github.com/riscv/riscv-isa-manual/issues/1397#issuecomment-2636906141) and the linked PR.

I would maybe revert #1852.

wmat, 2025-02-12:
@james-ball-qualcomm thoughts?  If you agree that we don't need this, I'll revert.  Please let me know.

james-ball-qualcomm, 2025-02-14:
You can revert it because Tim's tags.rb backend looks better. Tim, I left some review comments in PR #1769.

wmat, 2025-02-18:
OK thanks, 1852 was reverted.

PR #1852 (2025-02-11): Added Makefile target build-xml to build docbook XML format.
Fixes #1851