riscv/riscv-isa-manual #1854: Is there any possibility come lr-lr-sc scnerio from OS for one RV Hart?

omerkarslioglu, 2025-02-13:
Hi, without any aq and rl flag, is there any possibility come the instruction scnerio below?

```
lr (accesses x addr in mem)
lr (different from x addr)
sc (accesses x addr in mem)
```

omerkarslioglu, 2025-02-14:
Hi @allenjbaum you are right. Sc wil becomes failed, but my question is whether is there any possibility that comes any OS or program?

aswaterman, 2025-02-14:
There can be at most one outstanding reservation.  Therefore, the first LR is irrelevant.

The second LR might pair with the SC if their addresses are in the same reservation set (which, in many implementations, means the same cache line).  But this is not guaranteed.  Some implementations might deterministically fail the SC in this case, on the basis that the address differs from the most recent LR.

omerkarslioglu, 2025-02-15:
@aswaterman Thanks your answer but I am asking about that should I guarantee SC will fail in that scenario?

radimkrcmar, 2025-02-15:
> I am asking about that should I guarantee SC will fail in that scenario?

Your micro-architecture can always fail in this scenario, but without context, it's hard to tell whether it should (probably not).
Portable software must expect that this scenario will always fail.

Quoting the manual section "14.3. Eventual Success of Store-Conditional Instructions" (the definition with reservation sets in section 14.2 differs in `lr.d (x); sc.w (x+4)`, so it's better to start with 14.3):

> The Zalrsc extension defines constrained LR/SC loops, which have the following properties:
> * [...]
> * The SC must be to the same effective address and of the same data size as the latest LR executed
by the same hart.
>
> LR/SC sequences that do not lie within constrained LR/SC loops are unconstrained. Unconstrained
LR/SC sequences might succeed on some attempts on some implementations, but might never
succeed on other implementations.

aswaterman, 2025-02-16:
Yes, it is valid for your implementation to deterministically fail the SC if it's to a different address (regardless of whether it's in the same reservation set).  It's also legal for your implementation to perform the SC successfully if it's to a different address within the sam reservation set, but software can't rely on this behavior.

omerkarslioglu, 2025-02-18:
Thanks for your responses @aswaterman @radimkrcmar

closed 2025-02-18 by omerkarslioglu