riscv/riscv-isa-manual #1855: Cache policy attributes in RISC-V

oolegoon, 2025-02-13:
Hi! Can you please clarify is there some way to assign cache policy for virtual address in RISC-V like it is made in ARM?
I explored specs and didn't find anything. As I understand, current location for this can be custom PMAs. Am I right?

gfavor, 2025-02-13:
Take a look at the Svpbmt extension in the Priv manual.

closed 2025-02-13 by gfavor

oolegoon, 2025-02-13:
@gfavor , I read it. Svpbmt does not allow to specify write-back, write-through, read allocate, write allocate, etc.

gfavor, 2025-02-13:
Correct.  Svpbmt provides basic memory type overrides.  It does not provide the more extensive and granular specifying of individual attributes as is found in ARM architecture.

oolegoon, 2025-02-13:
And there is no extensions planned or in progress?

gfavor, 2025-02-13:
Also correct.  There has been no contingent of RVI members that have voiced a need to develop a standard for granular VA-based attribute overrides.  Presumably anyone that has desired this capability have implemented their own custom arch extension.