riscv/riscv-isa-manual #1859: ch6-sepmp --  Whether should we add a sfence.vma after configuring the epmp extension by setting the mseccfg.mmwp/mml bits?

rock-ifly, 2025-02-14:
Hi all, 
 
     As we know,   configuring the epmp extension by setting the mseccfg.mmwp/mml bits will change the usage of pmpcfg.LRWX bits, which seems to set a new PMP configuration.
 
    According to ch3.7 in RISCV manual volume II(privileged architecture), when PMP setting are modified, m mode software must use SFENCE.VMA x0, x0 instruction to synchronize the PMP setting with the virtual memory system.
 
    Then, 2 questions are here:
    1) Whether should we add a sfence.vma after configuring the epmp extension by setting the mseccfg.mmwp/mml bits?
 
    2) Is there any coding guide when we want to use the epmp extension? Such as choice 2.1 or choice 2.2 as following
    choice2.1)   
               a - set pmpcfg/pmpaddr
               b - sfence.vma x0, x0
               ......
               c -  set mseccfg.mmwp/mml
               d - sfence.vma x0, x0
               ......
    choice2.2) 
               a - set pmpcfg/pmpaddr
               b - sfence.vma x0, x0
                ......
               c -  set mseccfg.mmwp/mml
               d - set pmpcfg/pmpaddr    (reset pmp after start epmp extension)
               e - sfence.vma x0, x0
                ......
 
P.S. I have set a new topic in security group, but have on response, could anybody help us to fix this trouble? Thank you very much!

aswaterman, 2025-02-14:
The story is the same for Smepmp as it is for traditional PMP, and that story is described here: https://github.com/riscv/riscv-isa-manual/blob/70040578316b9978056c9f33ac654ea19f459169/src/machine.adoc#L3293-L3320

The TL;DR is that the SFENCE.VMA is needed if the implementation supports virtual memory, and it should be excluded otherwise, so as not to raise an exception on some implementations.

rock-ifly, 2025-02-16:
**So we SHOULD add a sfence.vma after configuring the epmp extension by setting the mseccfg.mmwp/mml bits for a page-based virtual memory system**, since   the story is the same for Smepmp as it is for traditional PMP.

Am I right?

> The story is the same for Smepmp as it is for traditional PMP, and that story is described here:
> 
> [riscv-isa-manual/src/machine.adoc](https://github.com/riscv/riscv-isa-manual/blob/70040578316b9978056c9f33ac654ea19f459169/src/machine.adoc#L3293-L3320)
> 
> Lines 3293 to 3320 in [7004057](/riscv/riscv-isa-manual/commit/70040578316b9978056c9f33ac654ea19f459169)
> 
>  [[pmp-vmem]] 
>  ==== Physical Memory Protection and Paging 
>   
>  The Physical Memory Protection mechanism is designed to compose with the 
>  page-based virtual memory systems described in 
>  <<supervisor>>. When paging is enabled, 
>  instructions that access virtual memory may result in multiple 
>  physical-memory accesses, including implicit references to the page 
>  tables. The PMP checks apply to all of these accesses. The effective 
>  privilege mode for implicit page-table accesses is S. 
>   
>  Implementations with virtual memory are permitted to perform address 
>  translations speculatively and earlier than required by an explicit 
>  memory access, and are permitted to cache them in address translation 
>  cache structures—including possibly caching the identity mappings from 
>  effective address to physical address used in Bare translation modes and 
>  M-mode. The PMP settings for the resulting physical address may be 
>  checked (and possibly cached) at any point between the address 
>  translation and the explicit memory access. Hence, when the PMP settings 
>  are modified, M-mode software must synchronize the PMP settings with the 
>  virtual memory system and any PMP or address-translation caches. This is 
>  accomplished by executing an SFENCE.VMA instruction with _rs1_=`x0` and 
>  _rs2_=`x0`, after the PMP CSRs are written. 
>  See <<hyp-mm-fences>> for additional synchronization requirements when the 
>  hypervisor extension is implemented. 
>   
>  If page-based virtual memory is not implemented, memory accesses check 
>  the PMP settings synchronously, so no SFENCE.VMA is needed. 
> The TL;DR is that the SFENCE.VMA is needed if the implementation supports virtual memory, and it should be excluded otherwise, so as not to raise an exception on some implementations.

aswaterman, 2025-02-17:
Yeah, that sounds right to me.

rock-ifly, 2025-02-21:
> Yeah, that sounds right to me.

Yes, so do I. Thank you!

Why not make it clear in the Sepmp extension or in the description of the mseccfg.mmwp/mml bits?

aswaterman, 2025-02-21:
I'll leave it to the Smepmp spec's authors, but a cross-reference to the section I quoted seems appropriate (but a duplication of the text is not a good idea).

mickflemm, 2025-03-03:
> > Yeah, that sounds right to me.
> 
> Yes, so do I. Thank you!
> 
> Why not make it clear in the Sepmp extension or in the description of the mseccfg.mmwp/mml bits?

Hello there, and sorry for not replying on the Security HC topic, seems I missed that.

To begin with, Smepmp is built on top of PMP, we didn't change the core mechanism and how the checks are performed, the priority/address matching etc, or any requirements related to virtual memory mappings to PMP regions and how PMP rules to such regions should be in-sync with the virtual memory subsystem. We just added a new set of access policies, mostly around M-mode. Unless you want to use a shared region between M-mode and S-mode under Smepmp, or used to access locked regions from S-mode under PMP, the encoding and behavior of e.g. a  R/W region accessible from S-mode is the same with PMP (pmpcfg.LRWX = 0110), and the same requirements apply regarding sfence.vma.

Smepmp basically adds 3 bits on mseccfg, and all of them are locked either on set (MMWP -now called MMAP- / MML), or on clear (RLB), they can't be modified afterwards and they are expected to be initialized asap during the early boot process / firmware initialization, long before any OS boots on S-mode and starts populating its page tables. At least that's the recommendation in case someone intends to use Smepmp, it's not expected to change machine mode access policies at some random point at runtime e.g. in parallel with page table updates. Therefore it didn't make much sense to say anything regarding sfence.vma in the description of mseccfg.mma/mml/rlb bits, it could even be misleading since those bits aren't meant to be modified when sfence.vma becomes relevant.