riscv/riscv-isa-manual #1866: RVA23 extensions with missing or incompete version numbers

tariqkurd-repo, 2025-02-18:
Most extensions are listed in the header of the unpriv or priv specs.
At least the following are missing from the headers:

Zcb
Zcmop
Zicbo[mpz]
Zkt
Zkr
Zkn
Zvfhmin
Zvfh
Zvkn[cg]
Zvks[cg]
Ssnpm
S[us]pm
Svbare/Sv39/Sv48/Sv57 (Note that other Sv extesnions like Svnapot _do_ have a version number).

The following extensions either don't have a listed version number, or in some cases more than one:

Zicbo[mpz]
Zicond - listed as 1.0 and 1.0.0
Zkt
Zkr - 1.0.1 in a changelog box only
Zbk[bc] are listed as frozen, ratified, 1.0 and 1.0.0-rc4
Zvfhmin
Zvfh
Zvkn[gc]
Zvks[cg]
Ssnpm
S[us]pm
Svbare/Sv39/Sv48/Sv57

I'm wondering if Svbare/Sv39 etc. are actually extensions, they are listed as such in the profiles.

gfavor, 2025-02-18:
Note that the Svbare/etc. "extensions" are examples of profile extension
names (versus actual arch extensions that we generally think in terms of)
that enable the profiles to refer to specific features within (in this
case) the Ss1p13 arch extension.

Generally most ratified extensions should be v1.0 (with rare exceptions
like E-Trace v2.0).  And there should only be x.y version numbers, i.e. no
x.y.z numbers.

Greg

aswaterman, 2025-02-18:
My recollection is that it is, by definition, unambiguous to refer to a ratified extension without a version number.  Doing so refers to what was ratified.

tariqkurd-repo, 2025-02-19:
Yeah thanks @gfavor , @aswaterman 
Note that the second list in my post above either have no version listed at all, or two different numbers.
There's no problem to refer to extensions without a number in general, but the ISA spec should specify exactly one version number for each extension.