riscv/riscv-isa-manual #1874: How does the privileged / unprivileged spec versioning work?

Timmmm, 2025-02-26:
I'm trying to add support for specifying a spec version to the Sail model, but I'm having a really hard time working out which versions of the specs actually exist. As far as I can tell there's

[RISC-V Technical Specifications](https://lf-riscv.atlassian.net/wiki/spaces/HOME/pages/16154769/RISC-V+Technical+Specifications) has the latest versions, Unpriv 20240411 and Priv 20240411

[RISC-V Technical Specifications Archive](https://lf-riscv.atlassian.net/wiki/spaces/HOME/pages/16154899/RISC-V+Technical+Specifications+Archive) has Unpriv 20191213 and Priv 20211203

These use date-based version numbers. However those surely aren't the only versions that have been released?

If you look in the latest documents there are sections titled:

Priv:

> Preface to Version 20240326
> Preface to Version 20240213
> Preface to Version 20211203
> Preface to Version 1.11
> Preface to Version 1.10
> Preface to Version 1.9.1

Unpriv:

> Preface to Document Version 20191213-Base-Ratified
> Preface to Document Version 20190608-Base-Ratified
> Preface to Document Version 2.2
> Preface to Document Version 2.1
> Preface to Version 2.0

But:

1. The 20240411 version of the priv spec starts with `Preface to Version 20240326`??
2. It looks like the priv spec switch from major.minor to dates after 1.11... but then the latest version talks about 1.14??

I am thoroughly confused. What released/ratified versions of the spec actually exist? Is there a list of them? Where can I download the PDFs? And what's with the random use of major.minor vs dates?

wmat, 2025-02-26:
Here's the current extension versioning schema:

[image]

gfavor, 2025-02-26:
On Wed, Feb 26, 2025 at 6:36 AM Bill Traynor ***@***.***>
wrote:

> So the dated specs are the "Published" integrated specs at a point in
> time. Each chapter is supposed to use the standard 0.1, 0.2, etc.
> versioning up to 1.0 which occurs at ratification. At that point, a PR is
> issued against the riscv-isa-manual repo to integrate the ratified ISA
> chapter. This occurs until the TSC(or ARC?) determines that it's time to
> publish a new dated release.
>
> Granted, this procedure has been somewhat loosely followed over the years
> with some content making it into riscv-isa-manual prior to ratification.
> There is an effort to pull out those bits, but that hasn't happened yet.
>
In response to a request, the ARC has discussed what should be the timing
and criteria for making official public ISA manual releases to the RISC-V
website.  A summary of that will appear in the next ARC minutes in the
coming week.

But in short the idea is that ratifications by the BoD (which can be at
most once a month) will trigger releases (once the ratified specs have been
properly merged); during long stretches of no ratifications the ARC will
identify when significant enough clarifications have merged into the
manuals such that it would be desirable to make a release; and AndrewW will
do some one-time work soon that will enable official public releases to
only include ratified specs (i.e. stripping out draft chapters).  The
desire is to have this process up and running before the European RISC-V
Summit.

Greg

Timmmm, 2025-02-26:
Ok so the components of the spec have major.minor versions, and then the overall spec has a dated version. That makes sense. But still are there really only two officially released versions of the overall specs?

And if I want to, say read the 1.13 version of the privileged architecture how do I know which date of the document to get?

And why does the 20240411 priv spec say "This release, version 20240213"?