riscv/riscv-isa-manual #1876: Exception code of PMBT=IO with misaligned access

Steven-Li-Xiaogang, 2025-02-28:
[image]

Privilege Specification has mentioned that PMBT field will override PMA memattr and cause page with stricter order. The IO feature with misaligned access should raise an exception. but it doesn't mentioned the exception code should be raised.

Should we raise an misaligned load/store exception? or we raise a load/store page fault cause it's caused by translation checking rules?

aswaterman, 2025-03-04:
You should raise whatever exception you would've raised if the PMA had indicated I/O.  The choice is implementation-defined, which is why we don't mandate a particular exception in the Svpbmt description.  (My recommendation is to raise an access exception, because that indicates the access should not be emulated by a misaligned address handler.)

Steven-Li-Xiaogang, 2025-03-05:
@aswaterman thanks a lot. closing this issue.

closed 2025-03-07 by Steven-Li-Xiaogang