riscv/riscv-isa-manual #1877: Question regarding phrasing for vector extension dependencies

nibrunie, 2025-03-02:
I have found, what I believe, are small discrepancies in the way we express extension "dependencies" in the vector extension specifications.
For example **Zvfbfmin**  is said to depends upon V or Zve32f while **Zvfhmin** is said to depend only upon Zve32f, see exact quote below:

> This extension depends upon either the "V" extension or the Zve32f embedded vector extension.

(source: https://github.com/riscv/riscv-isa-manual/blob/492f6e0c42f1310711dccf2ad2df95a4cad10c8c/src/bfloat16.adoc#zvfbfmin---vector-bf16-converts)

> The Zvfhmin extension depends on the Zve32f extension.

(source: https://github.com/riscv/riscv-isa-manual/blob/main/src/v-st-ext.adoc#zvfhmin-vector-extension-for-minimal-half-precision-floating-point)

I believe those extensions have exactly the same dependency scheme and if I am correct I think we should express it identically.
As far as I understand Zve32f is a "dependency" of "v", as "v" depends on "Zve64d" which in turns depends on "Zve64f" which depends on "Zve32f". And so I believe "Zvfbfmin" dependency scheme should be simplified to "Zve32f" without actually changing the specification intent nor meaning. 

Is my understand correct ?

nibrunie, 2025-03-02:
Similarly, I think we could clarify the vector crypto extensions dependencies

> The [Zvknhb](https://github.com/riscv/riscv-isa-manual/blob/main/src/vector-crypto.adoc#zvknh) and [Zvbc](https://github.com/riscv/riscv-isa-manual/blob/main/src/vector-crypto.adoc#zvbc) Vector Crypto Extensions --and accordingly the composite extensions [Zvkn](https://github.com/riscv/riscv-isa-manual/blob/main/src/vector-crypto.adoc#zvkn), [Zvknc](https://github.com/riscv/riscv-isa-manual/blob/main/src/vector-crypto.adoc#zvknc), [Zvkng](https://github.com/riscv/riscv-isa-manual/blob/main/src/vector-crypto.adoc#zvkng), and [Zvksc](https://github.com/riscv/riscv-isa-manual/blob/main/src/vector-crypto.adoc#zvksc)-- require a Zve64x base, or application ("V") base Vector Extension.
>
> All of the other Vector Crypto Extensions can be built on any embedded (Zve*) or application ("V") base Vector Extension.

(source: https://github.com/riscv/riscv-isa-manual/blob/492f6e0c42f1310711dccf2ad2df95a4cad10c8c/src/vector-crypto.adoc#extensions-overview)

- The "all the other vector crypto Extension" should depend on Zve32x, with the caveat that for Zvkb and Zvbb, the 64-bit version of the instructions is only supported on a Zve64x base.

aswaterman, 2025-03-04:
The spec says that V depends on Zve64d, Zve64d depends on Zve64f, and Zve64f depends on Zve32f.  So it suffices to express a dependence on only Zve32f.  Feel free to prepare a PR if you have time.

nibrunieAtSi5, 2025-03-04:
> The spec says that V depends on Zve64d, Zve64d depends on Zve64f, and Zve64f depends on Zve32f. So it suffices to express a dependence on only Zve32f. Feel free to prepare a PR if you have time.
> 
> (Likewise, Zve64d depends on Zve64x via Zve64f, so the crypto case you mention can be simplified, too.)

Sure: https://github.com/riscv/riscv-isa-manual/pull/1882