riscv/riscv-isa-manual #1880: what is the right VA width, 48-bit or 50-bit, for a system that supports Sv48/Sv48x4 mode?

rock-ifly, 2025-03-03:
what is the right VA width, 48-bit or 50-bit, for a system that supports Sv48/Sv48x4?

As the spec says, for a G-stage translation, the incomming IPA width shoud be 50-bit(48+2), as the following.
1) for a system with VS-stage on(!Bare, Sv48) and G-stage on(!Bare,Sv48x4), it is clear that VA width should be 48bit, according to Sv48.
2) but when it comes to VS-stage off(Bare) and G-stage on(Sv48x4), it seems VA width should to be 50bit for the system. 

Since the extra 2-bit matters for the TLB and Cache design, could anybody help me to fix the VA-width-selection problem for the Sv48x4 supporting system desing? Thanks.
 
> When hgatp .MODE specifies a translation scheme of Sv32x4, Sv39x4, Sv48x4, or Sv57x4, G-stage
address translation is a variation on the usual page-based virtual address translation scheme of Sv32,
Sv39, Sv48, or Sv57, respectively. In each case, the size of the incoming address is widened by 2 bits
(to 34, 41, 50, or 59 bits). To accommodate the 2 extra bits, the root page table (only) is expanded by a
factor of four to be 16 KiB instead of the usual 4 KiB. Matching its larger size, the root page table also
must be aligned to a 16 KiB boundary instead of the usual 4 KiB page boundary.

gfavor, 2025-03-03:
As the spec describes, valid VAs would be 48 bits and valid GPAs would be 50 bits.  If VS-stage is are, then VA=GPA and hence valid addresses are 50 bits.

Note that in probably many designs, the TLB holds complete VA-to-SPA translations and does not separately cache VA-to-GPA and GPA-to-SPA translations - in which case the larger GPA width only affects the tablewalk logic.

ved-rivos, 2025-03-03:
When VS-stage is Bare, the VA=GPA and is 50-bit wide. The TLB will need to support 50-bit wide VA tag to support Sv48x4 to be able to cache such VA-to-SPA translations.

gfavor, 2025-03-03:
Thanks for the correction as to the tagging of TLB entries.  Also note that one would also need to tag entries to distinguish whether they are caching a VA-to-SPA versus "bare VA/GPA-to-SPA" translation (or one would flush the TLB when satp.MODE is changed).

closed 2025-03-03 by gfavor