riscv/riscv-isa-manual #1881: How to deal with the address range overflows during memory access

qin996, 2025-03-03:
Should it raise an exception when VA overflows the max range of the VA?  For example,  VA is 64'hffff_ffff_ffff_fffff and access size is 2 bytes.
If raising an exception, the writing value of mtval is the instruction va(64'hffff_ffff_ffff_ffff) or the faulting va(may be 64'h0?)

gfavor, 2025-03-03:
VA address calculations don't "overflow", they just wrap around.  And if the second half of an access is the cause of the fault, then the VA of that second half is what is captured in *tval.

pdonahue-ventana, 2025-03-03:
This is described in the unpriv spec:
> The memory address space is circular, so that the byte at address is adjacent to the byte at address zero. Accordingly, memory address computations done by the hardware ignore overflow and instead wrap around modulo 2^XLEN.

closed 2025-03-03 by gfavor

qin996, 2025-03-04:
> VA address calculations don't "overflow", they just wrap around. And if the second half of an access is the cause of the fault, then the VA of that second half is what is captured in *tval.

If there is an address wrap, should an exception be generated? in my case, va=64'hffff_ffff_ffff_ffff，access size=2B.

unpriv spec describes that “The memory address space is circular, so that the byte at address is adjacent to the byte at address zero. Accordingly, memory address computations done by the hardware ignore overflow and instead wrap around modulo 2^XLEN.”

Is memory access permitted when an address wrap occurs?

qin996, 2025-03-04:
> VA address calculations don't "overflow", they just wrap around. And if the second half of an access is the cause of the fault, then the VA of that second half is what is captured in *tval.

Is memory access permitted when an address wrap occurs? If there is an address wrap, should an exception be generated? 

for example,
 1) if satp.mode = sv48, va=64'hffff_ffff_ffff_ffff，access size=2B, should the second half of this access  generate an exception due to address wrap? 
 2) if mmu is off, va = 64'h0000_ffff_ffff_ffff, and access size equal to 2B, the address of second half is 64'h0001_0000_0000_0000. Can it be considered that there has been an address wrap and an exception has occurred.

thanks

gfavor, 2025-03-04:
With Sv48, a VA that straddles the 64-bit wrap boundary (or is anywhere near it) will be an invalid address, As the arch spec says: "...  must have bits 63–48 all equal to bit 47, or else a page-fault exception will occur."

pdonahue-ventana, 2025-03-04:
@gfavor: Both 0xFFFFFFFFFFFFFFFF and 0x0000000000000000 are valid Sv48 addresses because both are sign-extensions of bit 47 (or bit 38 or any other bit).

On example 1, the spec doesn't say that there should be an exception on wraparound.  Therefore, there should be no exception.  The spec explicitly says that this should wrap around which means that it should successfully access those two bytes (similar to misaligned accesses that cross other page boundaries).

gfavor, 2025-03-04:
Thanks for the correction.  In typical Linux usage example 1 would be wrapping around from the top of kernel space into user space - which may or may not get caught by MMU permission checks (but won't be caught by the Sv48 check).  And note that the "wrap" point that the Sv48 check does catch (at 2^63-1) is when one "wraps" from the top of user space into kernel space.