riscv/riscv-isa-manual #1885: Does sfence.vma only effect page table whose vimd=0?

AlexGJL, 2025-03-04:
It seems that sfence.vma just affect page table whose vmid = 0,; hfence.vvma just affect page table whose vmid match hstap.vimd;  hfence.gvma just match vmid,  but I found below descrption in The RISC-V Instruction Set Manual: Volume II:
  " Similarly, if the setting of the PBMTE bit in menvcfg is changed, an HFENCE.GVMA instruction with rs1=x0 and rs2=x0 suffices to synchronize with respect to the altered interpretation of G-stage and VSstage PTEs' PBMT fields." why not use sfence.vma to flush page tables whose vmid equals 0 for upper case if there exist a V=0 to V=1 convert? 
  Does sfence.vma only effect page table whose vimd=0?

gfavor, 2025-03-04:
sfence.vma applies to caching of translations from the page tables pointed to by satp (or vsatp if V=1).  Vmid has no meaning for single-stage and VS-stage translations.  Also note that vmid=0 has no special architectural meaning versus vmid!=0 values.

aswaterman, 2025-03-04:
> Also note that vmid=0 has no special architectural meaning versus vmid!=0 values

I was going to chime in to say the same thing.  Our use of `x0` to indicate "all address spaces" can be confusing.  It doesn't mean that VMID 0 or ASID 0 is special; it means that the `x0` register specifier is special.

closed 2025-03-04 by gfavor

AlexGJL, 2025-03-05:
thank you, one more question:  HFENCE.GVMA can flush all G-stage or VS-stage page tables if rs1=x0, rs2=x0. Can HFENCE.GVMA flush VS-stage page tables if rs1=x0,rs2=!x0?

aswaterman, 2025-03-06:
Sure, it's always legal for an implementation to flush more than is strictly required.  (But software cannot rely on VS-stage translations being flushed.)

gfavor, 2025-03-06:
If a cached translation contains information from G-stage page tables, then it should be invalidated by an hfence.gvma.  And similarly, if a cached translation contains information from VS-stage page tables, then it should be invalidated by an hfence.vvma.

So a translation from a guest VA to a host PA would be affected by both forms of hfence.

AlexGJL, 2025-03-10:
ok, thanks for your reply.