riscv/riscv-isa-manual #1887: How to handle pc address out of VA/PA bound?

Steven-Li-Xiaogang, 2025-03-05:
A typical instruction which might be out of VA/PA bound is AUIPC which built a new pc address = PA + 32-bit u-immediate with lower 12 bits zeros. Assuming S/U mode is working at Sv48 scenario, PC with VA 0xffff_ffff_0000 and u-immediate of AUIPC is larger than 0x10000, then the new PC VA should be out of bound.
The question is: 
1. should hardware raise an exception immediately, or take no action and only do VA signed extended. 
2. if hardware takes no action, should compiler/software make hardware non running into this kind of issue?
3. if non of above, then the error will spread only other privilege/user/software running on the same core, how should we keep system to be RAS?

aswaterman, 2025-03-05:
AUIPC itself never raises exceptions as a result of its computation; it simply performs the computation `rd := pc + (imm20 << 12)` without regard to the meaning of the address.

If the result is later dereferenced or jumped to, then the implementation will raise an exception because the address is invalid.  But it is software's responsibility to make sure that addresses are valid so this situation does not arise.

Steven-Li-Xiaogang, 2025-03-07:
@aswaterman if the result is later jumped to, then the next instruction can validate this value (PC) as invalid.  But I think the new PC might be valid sometime even through virtual memory system is working (satp/vsatp/hgatp!=Bare), so would it be the way virus software can do attack?

lfiolhais, 2025-03-10:
What you are describing is a software attack trying to look for pages that are unprotected or shared with higher-privileged software. Whether some asset is protected or not is up to the software's security policy; it doesn't concern the ISA or the AUIPC wraparound as you described.

Steven-Li-Xiaogang, 2025-03-11:
@aswaterman @lfiolhais thank you for the clarification.
Another question about the AUIPC implementation: should hardware write wrapped value to rd when PC + (imm20 << 20) was out of valid VA/PA bound? should hardware do sign/zero expansion of the value written to rd?

Steven-Li-Xiaogang, 2025-03-28:
@aswaterman @lfiolhais thanks for clarification.

closed 2025-03-28 by Steven-Li-Xiaogang