riscv/riscv-isa-manual #1892: Can Store-Conditional retire if PMP/PMA checks would have caused an exception

Timmmm, 2025-03-06:
If PMP or PMA checks mean that a store conditional would raise an exception rather than store successfully, is it still allowed to retire and write a non-zero (failure) value to `rd`? In other words, do the PMP/PMA checks always have to happen first?

The only thing I can see in the spec about it is:

> For the purposes of memory protection, a failed SC.W may be treated like a store.

Which is super vague.

It also later says:

> Note that these definitions permit an implementation to fail an SC instruction occasionally for any reason

But it's not clear if that means it can fail for any reason *before* PMP/PMA checks or not.

I assume that PMP/PMA checks *must* happen first?

gfavor, 2025-03-06:
By definition an Access Fault (due to PMP/PMA checks) "faults" the SC instruction - which means that it is not retired and hence rd must not be modified.  But, since the architecture allows for occasional loss of a reservation, the attempted execution of the SC can still result in loss of the reservation.

Timmmm, 2025-03-06:
Right, but does the PMP/PMA check have to happen *first*, or is the SC allowed to retire with failure (non-zero value written to `rd`) even if it *would* have caused a fault had it continued?

aswaterman, 2025-03-06:
See https://github.com/riscv/riscv-isa-manual/pull/1239 for historical context.  It looks to me like the intent was to permit either behavior.  (The motivation was whether a PTE's D bit must, must not, or may be set on a failed SC, but the PMP/PMA story is closely related.)

gfavor, 2025-03-06:
I believe the former, i.e. if any load/store instruction fails MMU, PMP, or PMA permission checks, then the instruction faults - which means that it is not retired.  Whether an SC "fails" or not is just part of its execution behavior and is orthogonal to the permission checks that the instruction must pass.  

Put another way, there is nothing in the definition of SC that says that all of its permission checks apply only if the SC is going to succeed.

Regarding #1239, that is about whether side effects are allowed to happen even if an SC does execute and its execution behavior is to "fail".  But here we're talking about whether an SC is allowed to execute if it doesn't pass permission checks but is going to "fail".  And note that a "failing" SC changes arch state (i.e. rd) - which would be rather architecturally anachronistic to allow despite violating MMU, PMP, and/or PMA permission checks.

@aswaterman What are your thoughts now?

gfavor, 2025-03-06:
P.S. At the moment I'm not able to double-check my assertion  in the second paragraph of my last post.

pdonahue-ventana, 2025-03-06:
What if PMA checks are done out in the memory subsystem (e.g. the memory subsystem replies to the SC with an error if the address is read-only)?  The hart can locally determine some SC failures (SC following an SC or different address than the LR).  Does it still need to send the SC out do the PMA checks?

(This is similar to ARM's local monitor and global monitor.  I believe that sending it to the global monitor implies that it passed the local monitor.)

Timmmm, 2025-03-06:
Yeah that's what made me wonder. It might even be the case that if SCs are really allowed to fail for any reason then some implementations might just fail it without even bothering with the PMP check, even though that can be done locally.

I hope that isn't the case though. It feels like it should be that PMP checks *must* take priority but PMA checks may or may not (PMAs are almost entirely unspecified anyway so that's kind of implicit).

I'd be happy with a sentence like

> A failing store-conditional must still undergo memory protection checks and fault if it fails them. Only store-conditionals that pass memory protection checks can retire. 

Or something like that.

aswaterman, 2025-03-06:
@gfavor I suppose I'm second-guessing the wording we chose to resolve #1239.  Should we not have written "For the purposes of side effects of address translation and protection" instead of "For the purposes of memory protection" to make our motivations clearer?

In any case, I also have been operating under the assumption that the permissions checks are performed and actioned regardless as to whether the SC succeeds.

I'd propose replacing the problematic sentence with something like: "Memory permissions checks are performed for failed SC.W, but it is UNPREDICTABLE whether any side effects of address translation and protection are actioned."

Timmmm, 2025-03-06:
Does memory protection have any side effects?

aswaterman, 2025-03-07:
Setting the D bit in page-table entries, for example.

ved-rivos, 2025-03-07:
I also expected permission checks to be performed regardless of whether SC succeeds. The side effect of memory protection checks are the implicit loads and stores to the PTEs which may have a side effect of invalidating the reservation.

gfavor, 2025-03-07:
To Andrew in https://github.com/riscv/riscv-isa-manual/issues/1892#issuecomment-2705165532, I would agree.

As far as updated text, we don't have the architectural term UNPREDICTABLE, but we do have UNSPECIFIED.  So how about the following which includes another wording tweak and a parenthetical as well:

"Memory permissions checks are performed for a failed SC.W, but it is UNSPECIFIED whether any side effects of address translation and protection occur (such as setting a page table entry D bit)."

kasanovic, 2025-03-07:
I would try to be clearer that "No SC.W instruction shall retire unless it passes memory permission checks, but it is UNSPECIFIED whether any side effects of implicit address translation and protection memory accesses (such as setting a page table entry D bit) occur on a failed SC.W."

aswaterman, 2025-03-07:
I like that version better.

(I was consulting the Alpha ISA manual earlier today, which is probably why I wrote "UNPREDICTABLE"...)

gfavor, 2025-03-07:
Note that Krste's wording should not be specific or limited to the .W form of SC, i.e. just say "SC".

aswaterman, 2025-03-07:
It actually should be SC.W.  The way this section is written, the semantics of SC.W are described first, then SC.D is defined later to have the same behavior but for doublewords.

closed 2025-03-07 by aswaterman

Timmmm, 2025-03-07:
Thanks guys, that's much clearer!

PR #1893 (2025-03-07): Clarify that protection checks apply to failed SC
Resolves #1892 (#1239 redux)
commit 2c2a7938 (2025-03-07): Clarify that protection checks apply to failed SC (#1893)

Resolves #1892