riscv/riscv-isa-manual #1894: Should SFENCE.VMA and SFENCE.INVAL.IR ensure fetch after the two instruction will get new memory-management data structures

tom-guo-sudo, 2025-03-07:
1.	According to risc-v spec, if change memory-management data structures, instruction fetch should use new memory-management data structures after SFENCE.VMA and SFENCE.INVAL.IR.  Is this right?
_Executing an SFENCE.VMA instruction guarantees that any previous stores already visible to the current RISC-V hart are ordered before certain implicit references by subsequent instructions in that hart to the memory-management data structures._
_The SFENCE.INVAL.IR instruction guarantees that any previous SINVAL.VMA instructions executed by the current hart are ordered before subsequent implicit references by that hart to the memory-management data structures._

2. If item 1 is right, The two instruction need flush fetch pipeline to ensure this.  If memory-management data structures not influence instruction region, the flush will harm performance.
Can software determine whether to execute FENCE.I(flush pipeline) after a SFENCE.VMA/ SFENCE.INVAL.IR?

gfavor, 2025-03-08:
Items #1 and #2 are correct.  A design could avoid flushing the pipeline behind the SFENCE.VMA only if it could reliably determine that none of the following instructions that have already been fetched, don't reside in any of the pages affected by the SFENCE.VMA.  But practically speaking SFENCE.VMA is going to be a strongly synchronizing instruction that flushes the pipeline and doesn't restart/refill the pipeline until after all invalidations have been completed.

So yes, SFENCE.VMA instructions have a high execution "penalty"  Which is what motivated the creation of the SFENCE.INVAL instructions so that the synchronization costs can be amortized over several or many of these instructions when a batch of SFENCE's need to be performed.

Lastly note that there is no need for using a FENCE.I since the SFENCE.VMA and SFENCE.INVAL.IR instructions perform whatever synchronization is needed to implement the architecturally specified synchronization semantics.

closed 2025-03-08 by gfavor

aswaterman, 2025-03-08:
As an addendum to what Greg wrote, the reason FENCE.I is not needed in this use case is that instruction memory was never modified. In other words, FENCE.I handles changes to actual, physical instructions.