riscv/riscv-isa-manual #1896: mtimecmp is memory-mapped but stimecmp is not

omerguzelelectronicguy, 2025-03-10:
In the section of `stimecmp`, it is not mentioned that `stimecmp` is memory-mapped. However, `mtimecmp` is. I wonder the reason for chosing one of them is memory mapped and the other one not. In the riscv aclint spec, there is only mtimecmp but the memory region was 2-times of the `Machine-level Software Interrupt Device`. I thought `stimecmp` or the supervisor timer interrupts will be driven from aclint in the future release, just like mtimecmp is. Am I missing a specific reason?

aswaterman, 2025-03-10:
See the Sstc extension: https://github.com/riscv/riscv-isa-manual/blob/main/src/sstc.adoc

omerguzelelectronicguy, 2025-03-10:
@aswaterman Thanks for the answer. I already read that part of the spec (read again). I understand the reason for adding stimecmp. It is to have timer interrupt for supervisor level without the help of the mahcine mode timer interrupts. I didn't understand why it is a CSR rather than a memory-mapped register like mtimecmp. Is the STIMECMP register placed as CSR because CSR instructions are faster than memory instructions?

aswaterman, 2025-03-17:
CSR accesses aren't necessarily faster or slower than MMIO accesses, but IIRC part of the rationale here was making virtualization more straightforward. `stimecmp` accesses need to incorporate the `htimedelta` CSR and/or be intercepted by a hypervisor for emulation, both of which are easier if it's a CSR rather than a device.

omerguzelelectronicguy, 2025-03-24:
@aswaterman thank you for clarification.

closed 2025-03-24 by omerguzelelectronicguy