riscv/riscv-isa-manual #1898: when mcounter.tm = 0 and hvictl.vti = 1 then VS-mode write stimecmp will cause ill-inst-expt or vir-inst-expt?

noionion, 2025-03-12:
When mcounteren.tm = 0, a hart writing to stimecmp in VS-mode will trigger an illegal instruction exception.

If hvictl. VTI happens to be 1 during the access, according to the AIA specification, the hart should simultaneously generate a virtual instruction exception.

> In particular, if register vstimecmp is implemented (from extension Sstc), then attempts from VS-mode to write to stimecmp (or, for RV32 only, stimecmph) cause a virtual instruction exception when VTI = 1.

As described in the mcause section of the Privileged Spec 1.13, the exception priority should prioritize the illegal instruction exception in this case.  However, if the illegal instruction exception handler is delegated to VS-mode, VS-mode cannot resolve the issue because it lacks access to hvictl or mcounteren.  On the other hand, the virtual instruction exception can at most be delegated to HS-mode, which can address the problem caused by hvictl.

In this scenario, would it be more reasonable to prioritize the virtual instruction exception over the illegal instruction exception?

gfavor, 2025-03-13:
This excerpt from the H extension may help answer your question:

> When V=1, a virtual-instruction exception (code 22) is normally raised instead of an illegal-instruction 
exception if the attempted instruction is HS-qualified but is prevented from executing when V=1 either 
due to insufficient privilege or because the instruction is expressly disabled by a supervisor or 
hypervisor CSR such as scounteren or hcounteren.

noionion, 2025-03-13:
In this example, I specifically refer to mcounteren (not scounteren or hcounteren). The core issue here is:

When an instruction triggers two distinct exceptions due to conflicting controls from different CSRs, how should they be prioritized? Strict adherence to exception priority rules may leave software completely unable to resolve the situation.

noionion, 2025-03-13:
Thank you for the reply. I roughly understand now: When mcounteren.tm is set to 0, it inherently means that software does not intend for VS-mode to execute this instruction, and there is no need to resolve the resulting exception (as it stems from improper software implementation). Then raise a illegal instruction exception is right, even if the hardware implementation is likely to detect both types of exceptions.

closed 2025-03-13 by noionion