riscv/riscv-isa-manual #1909: Should we add an sfence.vma x0, x0 to synchronize after the PMA setting?

rock-ifly, 2025-03-17:
How to understand the following paragraph?
1) shoudld we add an sfence.vma x0,x0 to synchronize  after the PMA setting ? just like PMP setting
2) or should we must add a barrier for every PMA CSR setting , like flush the pipeline to make sure all the following ld/st instructions get the correct memory attributes ?

> Ch3.6.6 Coherence and cacheability PMAs says:
Where a platform supports configurable cacheability settings for a memory region, a platform-specific
machine-mode routine will change the settings and flush caches if necessary, so the system is only
incoherent during the transition between cacheability settings. This transitory state should not be
visible to lower privilege levels.

rock-ifly, 2025-03-17:
1) Is there only a common PMA setting for the whole SoC system?
2) If PE0 changes the PMA setting, must PE1 change the PMA setting accroding to PE? How PE0 let PE1 knows that the PMA setting is updated already?

allenjbaum, 2025-03-17:
IF PEx == HARTx:
The spec doesn't say anything about whether PMAs can be run time
configurable at all,
much less whether the configurable PMAs are shared between harts.
PMAs properties are speced  -- but their values, and (if configurable) how
they are configured, is completely implementation specific.
Thus, how PMAs can be configured and whether they are shared between HARTs
is custom to the implementation.

gfavor, 2025-03-17:
To the first questions, since the Priv spec doesn't define actual PMA CSRs (and associated read/write semantics), the spec doesn't say anything one way or the other about PMA change synchronization.  Put differently, PMAs are defined by the architetcure at a more conceptual level and the low-level ISA details (including synchronization requirements by M-mode firmware) are implementation specific.

rock-ifly, 2025-03-18:
@allenjbaum @gfavor Thanks for your reply.

Then PMAs properties are speced  -- but their values, and (if configurable) how they are configured, is completely implementation specific.
1) PMAs are defined by the architetcure at a more conceptual level 
2) and the low-level ISA details (including synchronization requirements by M-mode firmware) are implementation specific.

For me questions are still here,
1)  should the pmp/pma CSRs in a hart cover all the physical memory space? Every app can be mapped to different physical memory space, but the hart is able to access all the physical memory, isn't it?
2)  why not use the same PMA/PMP settings for all physical memory?  or why does different harts have different physical memory attributes?

gfavor, 2025-03-18:
All physical address space should be covered by PMAs and by PMPs.  With PMPs the spec says "If no PMP entry matches an M-mode access, the access succeeds", so by definition all of PA space will be covered.  A PMA implementation can choose to take an analogous approach where the default is that an address is inaccessible that doesn't match any PMA region.

Different regions of PA space have different characteristics, starting with main memory regions versus I/O regions.  Similarly, even with a main memory PMA region, M-mode firmware may want to protect part of main memory from non-M-mode access and only allow M-mode access.

Although some system designs may want some harts to have different PMA/PMP setting than other harts, many (or most?) system designs will generally configure all harts to have the same PMPs and PMAs.

aswaterman, 2025-12-09:
Closing, as Greg answered the original question.