riscv/riscv-isa-manual #1910: should we must add a barrier for every PMP CSR setting

rock-ifly, 2025-03-17:
should we must add a barrier for every PMP CSR setting , like flush the pipeline to make sure all the following ld/st instructions get the correct memory attributes ?

We get puzzled when the instruction flow in a superscalar processor is as the following 
......
csrrw pmpcfgX/pmpaddX
load  x5, mem[A]
sfence.vma x0, x0
......

**The load operation may be exectued speculatively, and used the pmp_cfg_OLD, not the expected pmp_cfg_NEW, except adding a barrier for every PMP CSR setting , like flush the pipeline to make sure all the following ld/st instructions get the correct memory attributes.**

Then how to deal with this?

> Ch3.7.2 Physical Memory Proctection and Paging
Implementations with virtual memory are permitted to perform address translations speculatively and
earlier than required by an explicit memory access, and are permitted to cache them in address
translation cache structures—including possibly caching the identity mappings from effective address
to physical address used in Bare translation modes and M-mode. The PMP settings for the resulting
physical address may be checked (and possibly cached) at any point between the address translation
and the explicit memory access. Hence, when the PMP settings are modified, M-mode software must
synchronize the PMP settings with the virtual memory system and any PMP or address-translation
caches. **This is accomplished by executing an SFENCE.VMA instruction with rs1= x0 and rs2= x0 , after
the PMP CSRs are written**. See Section 21.5.3 for additional synchronization requirements when the
hypervisor extension is implemented

rock-ifly, 2025-03-17:
1) Is there only a common PMP setting for the whole SoC system? that is for all the PEs, there is only one PMP setting, the same PMP setting?
2) If PE0 changes the PMP setting, must PE1 change the PMP setting accroding to PE0? How PE0 let PE1 knows that the PMP setting is updated already?

allenjbaum, 2025-03-17:
My understanding:
If PEx == HARTx, and "common" means that setting a PMP CSR in one hart
affects the PMP CSR in another,
then the answer is no:  CSRs are private to each HART (possibly with
exceptions for some specific examples of counter CSRs).
It is up to Mmode firmware to communicate any required change between harts
(e.g. with software interrupts)

gfavor, 2025-03-17:
To the first question, the key hint in the spec if the sentence preceding the highlighted sentence, i.e. "Hence, when the PMP settings are modified, M-mode software must synchronize the PMP settings with the virtual memory system and any PMP or address-translation caches."  This indicates that by default PMP settings (in "Implementations with virtual memory") are not synchronized wrt surrounding loads/stores.

aswaterman, 2025-03-17:
> This indicates that by default PMP settings (in "Implementations with virtual memory") are not synchronized wrt surrounding loads/stores.

And, conversely, PMP settings take effect immediately in systems without virtual memory.

rock-ifly, 2025-03-18:
> My understanding:
> If PEx == HARTx, and "common" means that setting a PMP CSR in one hart
> affects the PMP CSR in another,
> then the answer is no:  CSRs are private to each HART (possibly with [...]

Yes, PEx=HARTx.
So CSRs are private to each HART, then every hart has his own CSRs, and uses Mmode firmware to make sure all harts in the SoC works in the same CSR setting by softwre intterrupts.

Thanks a lot.

rock-ifly, 2025-03-18:
> > This indicates that by default PMP settings (in "Implementations with virtual memory") are not synchronized wrt surrounding loads/stores.
> 
> And, conversely, **PMP settings take effect immediately in systems without virtual memory.**

**PMP settings take effect immediately in systems without virtual memory.** This means  **a barrier is a MUST** for PMP CSR setting.

Or the load operation may be exectued speculatively, and used the pmp_cfg_OLD, not the expected pmp_cfg_NEW, ---> PMP settings don't  take effect immediately in systems without virtual memory, like m mode.

gfavor, 2025-03-18:
The spec reference in my last post is addressing the possibility that hardware may be caching in a TLB not just the VA-to-PA translation and associated permission information, but also PMP-related permission information.  This concern is separate from software changing the page tables and needing to synchronize those changes with a suitable form of sfence.

In short, just like with caching of translations in a TLB, caching of PMP information in a TLB requires use of an sfence to invalidate any/all TLB entries that may be affected by a PMP change.

gfavor, 2025-03-18:
Yes, in systems without virtual memory, the normal CSR synchronization rules apply - which mean that the very next instructions must see the effect of changed PMP settings.  Any speculative execution of these following instructions will need to be thrown away and re-executed after the new PMP settings are in place.

rock-ifly, 2025-03-18:
@gfavor Thanks a lot. That is very clear.

closed 2025-03-18 by gfavor