riscv/riscv-isa-manual #1912: should all the explict CSR read/write operation trigger a micro-arch implemented barrier, or followed a FENCE/Sfence operation

rock-ifly, 2025-03-17:
1) should all the explict CSR read/write operation be followed by a synchronize or barrier operation?
2) how does the synchronize or barrier operation implemented? 
2a) micro-arch implemented barrier, such as description in Paragraph-1 
      --- trigger a micro-arch implemented barrier? like flush the pipeline to make sure all the following instructions can get/use the correct NEW csr vaule?
2b) software routine to implement a FENCE/Sfence operation, to enforce order in all case,such as  description in Paragraph-3
      --- such as sfence .vma x0 x0 after pmp setting
3) what should we take to make sure a correct explict CSR read/write? which one is better?
3a) either 2a or  2b ?
3b) both 2a and 2b ?

> volI-unprivileged-ch7.1.1
**Each RISC-V hart normally observes its own CSR accesses, including its implicit CSR accesses, as
performed in program order. In particular, unless specified otherwise, a CSR access is performed after
the execution of any prior instructions in program order whose behavior modifies or is modified by
the CSR state and before the execution of any subsequent instructions in program order whose
behavior modifies or is modified by the CSR state.** Furthermore, an explicit CSR read returns the CSR
state before the execution of the instruction, while an explicit CSR write suppresses and overrides any
implicit writes or modifications to the same CSR by the same instruction.
Likewise, any side effects from an explicit CSR access are normally observed to occur synchronously
in program order. Unless specified otherwise, the full consequences of any such side effects are
observable by the very next instruction, and no consequences may be observed out-of-order by
preceding instructions. (Note the distinction made earlier between side effects and indirect effects of
CSR writes.)
For the RVWMO memory consistency model (Chapter 18), CSR accesses are weakly ordered by default,
so other harts or devices may observe CSR accesses in an order different from program order. In
addition, CSR accesses are not ordered with respect to explicit memory accesses, unless a CSR access
modifies the execution behavior of the instruction that performs the explicit memory access or unless
a CSR access and an explicit memory access are ordered by either the syntactic dependencies defined
by the memory model or the ordering requirements defined by the Memory-Ordering PMAs section in
Volume II of this manual. **To enforce ordering in all other cases, software should execute a FENCE
instruction between the relevant accesses.** For the purposes of the FENCE instruction, CSR read
accesses are classified as device input (I), and CSR write accesses are classified as device output (O).

aswaterman, 2025-03-17:
> should all the explict CSR read/write operation be followed by a synchronize or barrier operation?

No, because of the first two sentences you quoted: "Each RISC-V hart normally observes its own CSR accesses, including its implicit CSR accesses, as performed in program order. In particular, unless specified otherwise, a CSR access is performed after the execution of any prior instructions in program order whose behavior modifies or is modified by the CSR state and before the execution of any subsequent instructions in program order whose behavior modifies or is modified by the CSR state."

> how does the synchronize or barrier operation implemented?

Therefore, it is the microarchitecture's responsibility to handle these cases (unless otherwise specified, e.g. for page-table modifications, where software must execute an SFENCE.VMA instruction).

> what should we take to make sure a correct explict CSR read/write? which one is better?

(2a) is required (unless otherwise specified), so there isn't much of a choice here.

rock-ifly, 2025-03-18:
@aswaterman Thanks, then the microarchitecture takes the responsibility to 
1)implement a necessrary barrier for some CSRs setting, like pmp setting which need to make sure that "PMP settings take effect immediately in systems without virtual memory", or 
2) to build a routine for software, like sfence.vma after page-table modification（satp CSR update）

gfavor, 2025-03-18:
In short, that is correct.  Unless the architecture places responsibility on software to use sfence instructions to synchronize MMU/PMP chnages with hardware caching structures, the default CSR ordering rules apply - which say that the effect of a CSR write must be seen by all following instructions in program order.

closed 2025-12-09 by aswaterman