riscv/riscv-isa-manual #1915: how to deal with pointer masking when vsstatus.MXR or sstatus.MXR is set

qin996, 2025-03-19:
In pointer masking spec, it describes that _"pointer masking does not apply when MXR is set."_

However, there are vsstatus.MXR and sstatus.MXR when V=1, and Privileged spec describes that _"The vsstatus field MXR, which makes execute-only pages readable by explicit loads, only overrides VS-stage page protection. Setting MXR at VS-level does not override guest-physical page protections. Setting MXR at HS-level, however, overrides both VS-stage and G-stage execute-only permissions."_

So, for example,

1）When one of vsstatus.MXR or sstatus.MXR is set, of course, V=1 and vsatp.mode is not bare, should pointer masking be applied?
2）And when V=1, vsatp.mode is bare. Should we ignore vsstatus.MXR and only focus on sstatus.MXR for applying pointer masking?

I ask the same questions in riscv-j-extension, but nobody response.

thanks

gfavor, 2025-03-19:
I'm forwarding this to Martin and the J group.  Since I can't see your full email address, I can't cc you.  But this will appear on the J group's email list, so you can see it there.

radimkrcmar, 2025-03-19:
I think the sentence before your quote limits the scope to MPRV/SPVP:
> MPRV and SPVP affect pointer masking as well, causing the pointer masking settings of the effective
privilege mode to be applied.  When MXR is in effect at the effective privilege mode where explicit
memory access is performed, pointer masking does not apply.

The idea is that pointer masking can be ignored when a higher privilege level accesses memory though MPRV (and H[LS]V).

Accesses done by the actual V=1 execution should always be subject to pointer masking:
1. Yes.
2. Apply pointer masking regardless of sstatus.MXR.

A convoluted case is V=0, MPRV=1, SPVP=1, vsapt.mode=bare, sstatus.MXR=0, and vsstatus.MXR=1.
vsstatus.MXR is not in effect, so pointer masking would apply.

viktoryou, 2025-03-20:
Need help to clear my mind of this special MXR mechanism.
The Priv spec said _MXR has no effect when page-based virtual memory is not in effect._. While the Pointer masking spec said _although MXR has no effect on permissions checks when page-based virtual memory is not in effect, it is still used in determining whether or not pointer masking should be applied._

For the following case, whether pointer masking should be applied:
(a) current is M-mode, effective_virt=0, either sstatus.mxr or vsstatus.mxr is set
(b) current_virt=0, effective_virt=1(changed by H[LS]V), vsatp.mode and hgatp.mode are bare, either sstatus.mxr or vsstatus.mxr is set
(c) current_virt=0, effective_virt=1(changed by H[LS]V), vsatp.mode is non-bare, either sstatus.mxr or vsstatus.mxr is set
(d) current_virt=1, effective_virt=1, either sstatus.mxr or vsstatus.mxr is set

These case come to my mind and start to become confusing. 
> The idea is that pointer masking can be ignored when a higher privilege level accesses memory though MPRV (and H[LS]V).

@radimkrcmar Seems H[LS]V also places similar limitation on the case where pointer masking does not apply when MXR is set, implicitly. Should these cases take into consideration whether page-based virtual memory is in effect?

> Accesses done by the actual V=1 execution should always be subject to pointer masking

What does actual V=1 mean?  Is the case where effective_virt is same with current_virt?

radimkrcmar, 2025-03-20:
> Need help to clear my mind of this special MXR mechanism. The Priv spec said _MXR has no effect when page-based virtual memory is not in effect._. While the Pointer masking spec said _although MXR [...]

Good point.  I am sorry that my last example added to the confusion.

As I interpret the spec, MXR is in effect with BARE translation mode for the purposes of MPRV/H[LS]V (MXR just has no effect on permission checks).

> For the following case, whether pointer masking should be applied:
> (a) current is M-mode, effective_virt=0, either sstatus.mxr or vsstatus.mxr is set

If mstatus.MPRV and mstatus.MXR (= sstatus.MXR) are set, then accesses are not subjected to pointer masking.  satp.mode doesn't make a difference.  vsstatus.MXR isn't looked at.

> (b) current_virt=0, effective_virt=1(changed by H[LS]V), vsatp.mode and hgatp.mode are bare, either sstatus.mxr or vsstatus.mxr is set
> (c) current_virt=0, effective_virt=1(changed by H[LS]V), vsatp.mode is non-bare, either sstatus.mxr or vsstatus.mxr is set

(As I mentioned above, hgatp.mode and vsatp.mode should be irrelevant.)

I think that vsstatus.MXR doesn't affect pointer masking in (b) and (c).
The reason is that in case of sstatus.MXR=0 and vsstatus.MXR=1, the higher privilege level would have to resort to full software emulation if it wanted to correctly perform the guest access with protections when the guest translation depends on MXR for some reason.
The whole point of H[LS]V is to avoid that.

Please don't take it as the truth, I am always grasping at straws just like you are.

> (d) current_virt=1, effective_virt=1, either sstatus.mxr or vsstatus.mxr is set

MPRV=0, because it must be reset on mret/sret to the less privileged mode, so pointer masking applies.

> These case come to my mind and start to become confusing.

I agree that the spec should be clearer on this.

> > The idea is that pointer masking can be ignored when a higher privilege level accesses memory though MPRV (and H[LS]V).
> 
> [@radimkrcmar](https://github.com/radimkrcmar) Seems H[LS]V also places similar limitation on the case where pointer masking does not apply when MXR is set, implicitly. Should these cases take into consideration whether page-based virtual memory is not in effect?

No.  H[LS]V should behave the same as MPRV with MPV, but I don't think that translation mode makes a difference for pointer masking.
 
> > Accesses done by the actual V=1 execution should always be subject to pointer masking
> 
> What does actual V=1 mean? Is the case where effective_virt is same with current_virt?

Yes, the hart is actually in VS or VU mode.

qin996, 2025-03-24:
@gfavor Hi, regarding this issue, if the group has already decided what to do, please reply here. Thank you very much.

gfavor, 2025-03-24:
One of the chairs of the J group posted the following on Friday: "Thanks Greg. We will cover this topic on the next J-extension meeting and let you know what the group decides."

So I suggest attending the next J meeting to hear the discussion and the conclusion that is reached (as well as have the chance to ask any clarifying questions).

gfavor, 2025-03-24:
Courtesy of Ved, I believe the following addresses the two questions originally posed:

The specification states:
     "When MXR is in effect at the effective privilege mode
      where explicit memory access is performed, pointer
      masking does not apply."
and further clarifies:
    "Note that this includes cases where page-based virtual
     memory is not in effect; i.e., although MXR has no effect
     on permissions checks when page-based virtual memory is
     not in effect, it is still used in determining whether or
     not pointer masking should be applied."

To determine whether MXR is in effect, the rule in Table 42
should guide.
1. If the effective privilege mode of the access is VS or VU,
    then MXR is in effect if either vsstatus.MXR is 1 or
    sstatus.MXR is 1.
2. If the effective privilege mode of the access is S/HS, then
    MXR is in effect if sstatus.MXR is 1.

On the two questions asked in GitHub, I answer them as follows:
"1）When one of vsstatus.MXR or sstatus.MXR is set, of course, V=1 and
     vsatp.mode is not bare, should pointer masking be applied?"
Pointer masking is not applied. The status of vsatp.MODE is immaterial
as noted by the pointer masking specification.

"2）And when V=1, vsatp.mode is bare. Should we ignore vsstatus.MXR and
    only focus on sstatus.MXR for applying pointer masking?"
For explict accesses to effective privilege mode VS/VU has MXR
in effect if sstatus.MXR is 1 or if vsstatus.MXR is 1.

radimkrcmar, 2025-03-25:
Thanks to Ved for clearing this up.                                                                                                                                                       
                                                                                                                                                                                               
I think the manual could mark the sentence about MPRV as non-normative to make it clearer that it is just stating the obvious:                                                                                                                                                                               
                                                                                                                                                                                               
 >MPRV and SPVP affect pointer masking as well, causing the pointer masking settings of the effective privilege mode to be applied.

I assume all normative text is there for a reason, and that the sentence was somehow modifying the meaning of the next sentence, because the correct interpretation weakens the security guarantees provided by pointer masking for S-mode.

qin996, 2025-03-31:
Thanks all for clarification, however, if mstatus.MXR=1, should pointer masking be applied in m mode?
according to ved's reply https://github.com/riscv-software-src/riscv-isa-sim/issues/1940#issuecomment-2761370639 , mstatus.MXR does not affect memory accesses made with effective privilege mode equal to M.
and point mask spec descirbes that

> When MXR is in effect at the effective privilege mode where explicit memory access is performed, pointer masking does not apply.

> Note that this includes cases where page-based virtual memory is not in effect; i.e., although MXR has no effect on permissions checks when page-based virtual memory is not in effect, it is still used in determining whether or not pointer masking should be
applied.

gfavor, 2025-03-31:
The middle of https://github.com/riscv/riscv-isa-manual/issues/1915#issuecomment-2749578594 effectively answers the question, i.e. mstatus.MXR is never in effect for M-mode.

closed 2025-04-04 by gfavor