riscv/riscv-isa-manual #1919: All traps at any privilege level are handled in machine mode?

kdockser, 2025-03-20:
In the privileged architecture, section **3.1.8 Machine Trap Delegation (medeleg and mideleg) Registers** the first sentence declares:

`By default, all traps at any privilege level are handled in machine mode, though a machine-mode handler can redirect traps back to the appropriate level with the MRET instruction`

However, **11.1.3. Supervisor Interrupt (sip and sie) Registers** states:

`An interrupt i will trap to S-mode if both of the following are true: (a) either the current privilege mode is S and the SIE bit in the sstatus register is set, or the current privilege mode has less privilege than S-mode; and (b) bit i is set in both sip and sie.`

1. It appears that we have a contradiction as clearly it is not the cause that by default all traps at any privilege level are handled in machine mode. Is this text out of date?
2. Does `MRET` redirect traps from an M-mode trap handler to an S-Mode trap handler for S-Mode interrupts (for example)?

aswaterman, 2025-03-20:
"By default" needs to be read in the context of the entire paragraph.  The next sentence describes the exception- and interrupt-delegation mechanisms: that is to say, the default out of reset is that all traps go to M-mode, but delegation can be configured to send traps to S-mode.

The supervisor text you quoted only applies to traps that have been delegated to S-mode--otherwise they wouldn't appear in `sip`--and so there isn't actually a contradiction.

kdockser, 2025-03-20:
Thanks Andrew. I was thinking about an external interrupt controller writing mip.SEIP and forgot that delegation is required to make this visible in sip.

closed 2025-03-20 by kdockser

aswaterman, 2025-03-20:
No prob, happy to help.