riscv/riscv-isa-manual #1924: Questions about RVV 1.0 Post-Ratification Changes

QJtaibai, 2025-03-24:
We have noticed that since the ratification of RVV 1.0 on August 8, 2023, there have been numerous additional modifications.

This raises some concerns regarding the implementation of compatibility tests for RVV 1.0 moving forward, because it's still called RVV1.0 on spec.(https://github.com/riscv/riscv-isa-manual/blob/main/src/sstc.adoc)

Concerns:
Post-Ratification Changes:
Many pieces of hardware produced before August 8, 2023, which passed RVV 1.0 compatibility testing at that time, are unable to support the current version of the specification (the so-called rvv1.0).
Hardware manufactured today may also quickly become outdated as the specification continues to evolve, despite initially being compliant.
Suggested Actions:
In light of these issues, we would like to propose the following:

Exclusion of Post-Ratification Changes:
Ensure that all subsequent modifications after August 8, 2023, are explicitly excluded from RVV 1.0 compatibility testing.
Version Number Adjustment:
Amend the specification to either remove the version number reference or update it to reflect the changes appropriately.
For example, the version number could be updated to RVV 1.0.1 to differentiate it from the ratified RVV 1.0.
These proposed steps aim to maintain the integrity of compatibility tests for RVV 1.0 and clearly separate the original ratified version from any subsequent revisions.

We believe these actions will help ensure the robustness and reliability of RVV 1.0 compatibility tests, both for existing hardware and future productions.

camel-cdr, 2025-03-24:
What changes did you notice?

From what I can tell, everything is just formattibg/wording changes and clarifications.

I suppose [this](https://github.com/riscv/riscv-isa-manual/commit/6cd53fe0302ffeccd761b55919bad605d36cc04e) may technically be a breaking change, but I doubt anyone implemented a behavior not compatible with this typo fix. Just looking at the changed segment, you could've techincally implemented vslideup/vslidedown to change the first element after a context switch, but this would've made it basically impossible to write code using these instructions, and it goes against all prior definitions of vstart behavior.

There also were a few changes based on [this vector SIG discussion](https://lists.riscv.org/g/tech-vector-ext/message/845), but:

> The proposed solution defines this case as always agnostic so existing
implementations can continue to work as before (e.g., implementing
undisturbed when requested), while not burdening implementations that
rearrange data internally.

QJtaibai, 2025-03-24:
[image]
I have identified a potential issue with a recent modification and would like to clarify its impact on the vmvnr instruction, specifically when the VILL bit is set to 1.
Is the current design intended to reserve the instruction's behavior when the VILL bit is 1?
Should an illegal instruction exception be recommended or preferred in this scenario instead? The ratified RVV1.0 design does not create an exception when vill is set.

Thanks

topperc, 2025-03-24:
> <img alt="Image" width="844" src="https://private-user-images.githubusercontent.com/117722273/426050731-6777ce9d-990e-4f9b-8138-aa9154576641.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDI4MzkxNjYsIm5iZiI6MTc0MjgzODg2NiwicGF0aCI6Ii8xMTc3MjIyNzMvNDI2MDUwNzMxLTY3NzdjZTlkLTk5MGUtNGY5Yi04MTM4LWFhOTE1NDU3NjY0MS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMzI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDMyNFQxNzU0MjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xYzQzOGFhZTdlNGJjNWUxODhiNGNkN2UzMWI2MjczMjczYThmZTEzYTU3MTY0ZDcxZjc2ZDNkYzVhYTZkODY2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.6KbiAME-IXUxd19JDaQvWinhmG8TDAOn_e5yoeSo2Ys"> I have identified a potential issue with a recent modification and would like to clarify its impact on the vmvnr instruction, specifically when the VILL bit is set to 1. Is the current design intended to reserve the instruction's behavior when the VILL bit is 1? Should an illegal instruction exception be recommended or preferred in this scenario instead? The ratified RVV1.0 design does not create an exception when vill is set.
> Thanks

Whole register move was made dependent on `vtype` before 1.0 ratification. https://github.com/riscvarchive/riscv-v-spec/commit/f1d349d79d0a14ed0d085c469f9032f2eb38fa6d Unfortunately, the sentence you highlighted was missed in that update.

closed 2025-03-31 by QJtaibai