riscv/riscv-isa-manual #1925: TLB causes inconsistent behavior when svadu is implementated

leBsky, 2025-03-24:
[image]
Spec Step 9 claims that each translation should update A/D bits if needed.

However, assume TLB caches the PMP flag, then the following instruction sequence may cause inconsistent behavior with Spec:
1. Load access to a page whose pte.AD are 2'b10 and  pmp flag xwr==001 (read only), then this entry is cached in TLB.
2. Store access to the same page and finds pmp check violation, then raise access fault directly without updating the associated D bit in pte. Is it allowed or the store access here must update D bit anyway?

leBsky, 2025-03-24:
> Existence of TLBs must not change the exception priority specified in Table 34.

Access fault will be triggered anyway in both case, the point is the A/D bit will not be updated due to the use of TLB.  And I am not sure if this behavior is legal.

ved-rivos, 2025-03-26:
Sorry, my point was that the architectural behavior should not change based on existence of TLBs.

ved-rivos, 2025-04-17:
Hope that helped. Please close the issue if there are no further questions

closed 2025-04-18 by leBsky